ARMmbed/mbed-os #1666: MBED hangs during bootloader program jump while using rtos

pi19404, 2016-04-13:
Hi

i am using MBED to implement a bootloader based application on NUCLEO_091RC board and STM32F030CC controller also.

The application works correctly when i do not use RTOS in my main application.
When i include RTOS in my application,after program jump the kernel hangs

If i do not perform any program jumps then RTOS works well in the bootloader application also

using GDB i found that program enters WWDG_IRQHandler () just after execution of the below function __svcKernelStart () at /mnt/hgfs/repos/mbed/libraries/rtos/rtx/TARGET_CORTEX_M/rt_CMSIS.c:414

__svcKernelStart () calls the svcKernelStart () function in privileged mode
and code for that is found at line 198 of mbed/libraries/rtos/rtx/TARGET_CORTEX_M/rt_CMSIS.c file

#define SVC_Call(f)                                                            \
  __asm volatile                                                                 \
  (                                                                            \
    "ldr r7,="#f"\n\t"                                                         \
    "mov r12,r7\n\t"                                                           \
    "svc 0"                                                                    \
    :               "=r" (__r0), "=r" (__r1), "=r" (__r2), "=r" (__r3)         \
    :                "r" (__r0),  "r" (__r1),  "r" (__r2),  "r" (__r3)         \
    : "r7", "r12", "lr", "cc"                                                  \
  );

below is a test program to simulate the issue :
i jump to the same default program address 0x08000000 in and RTOS thread

https://gist.github.com/pi19404/928e024cd8008e814357738877cfd67e

0xc0170, 2016-04-13:
kernelStart is invoked prior main, thus I don't fully understand the flow of your program. 

> When i include RTOS in my application,after program jump the kernel hangs

Can you share the callstack? what does it mean program jump? Which line in your code gist?

pi19404, 2016-04-14:
The application i was writing is a bootloader program where in program at
0x08000000 contains a bootloader which either upgrades the main application program at address 0x0800C800 or jump to that location to start executing the main application

The  application address APPLICATION_ADDRESS is configured in my actual application as 0x0800c800.When i do not use RTOS calls in my application,The program runs fine,the application jumps from bootloader to the main application at 0x0800c800 and starts executing the code.

In order to debug whether the issue was due to jump to 0x0800c800 ( which would involve changes in linker script and other files)  or RTOS . I set the application address as the 0x08000000 but still observed the same issue. Indicating that issue lies with the RTOS calls and not due to the 

`jump_application()` function is responsible for setting the program counter to the new address and then start executing the code located at address specified at `APPLICATION_ADDRESS`

`__set_MSP(*(__IO uint32_t*)APPLICATION_ADDRESS);`  Sets Main Stack Pointer.
while the program jumps and starts executing code from the new flash address `execute_application();`
is called.

However when i configure RTOS thread then ,When program jump is triggered by calling address execute_application(); . 

The main program enters the `_svcKernelStart ()` and then enters `WWDG_IRQHandler` and loops indefinitely when executing the below function call

```
SVC_0_1(svcKernelStart,      osStatus, RET_osStatus)

which is defined as

#define SVC_0_1(f,t,rv)                                                        \
__attribute__((always_inline))                                                 \
static inline  t __##f (void) {                                                \
  SVC_Arg0();                                                                  \
  SVC_Call(f);                                                                 \
  return (t) rv;                                                               \
}
```
which defines a function `__svcKernelStart` which calls the `SVC_Call` function with `svcKernelStart` as argument.
```
#define SVC_Call(f)                                                            \
  __asm volatile                                                                 \
  (                                                                            \
    "ldr r7,="#f"\n\t"                                                         \
    "mov r12,r7\n\t"                                                           \
    "svc 0"                                                                    \
    :               "=r" (__r0), "=r" (__r1), "=r" (__r2), "=r" (__r3)         \
    :                "r" (__r0),  "r" (__r1),  "r" (__r2),  "r" (__r3)         \
    : "r7", "r12", "lr", "cc"                                                  \
  );
```

Yes The `kernelStart` is invoked prior to the main and it thus enters the `WWDG_IRQHandler` and loops indefinitely and does not enter the main program at all.This behavior occurs only after the program jump .

dadiliebian, 2017-04-12:
Hello,I have a question. I make the bootloader 's .bin and combine with application.bin.The bootloader's function is jump directly application.But it does not.
The follow is bootloader code.

#include <events/mbed_events.h>
#include <mbed.h>

int main()
{
mbed_start_application(POST_APPLICATION_ADDR);
}

0xc0170, 2017-04-12:
@dadiliebian Please create a new issue for your use case. And describe it there in details for us to be able to reproduce. What target, toolchain, what commands, examples are you running? Thanks

0xc0170, 2017-04-12:
@pi19404 Have you resolved your issue? We added a bootloader support , that might resolve this issue

martinichka, 2018-05-03:
@pi19404 I had similar behavior, but fixed it. Did you update SCB->VTOR correctly, see patch below

```diff --git a/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/system_clock.c b/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/system_clock.c
index 18e823a..22071cd 100644
--- a/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/system_clock.c
+++ b/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/system_clock.c
@@ -31,6 +31,7 @@
 
 #include "stm32f4xx.h"
 #include "mbed_assert.h"
+#include "nvic_addr.h"
 
 /*!< Uncomment the following line if you need to relocate your vector Table in
      Internal SRAM. */
@@ -94,7 +95,7 @@ void SystemInit(void)
 #ifdef VECT_TAB_SRAM
     SCB->VTOR = SRAM_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
 #else
-    SCB->VTOR = FLASH_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal FLASH */
+    SCB->VTOR = NVIC_FLASH_VECTOR_ADDRESS; /* Vector Table Relocation in Internal FLASH */
 #endif
 
 }
`

cmonr, 2018-06-04:
@ARMmbed/team-st-mcd

0xc0170, 2018-06-05:
> @ARMmbed/team-st-mcd

just one question - does  `targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/system_clock.c` the file needs any editing as proposed above? If not, we can close this issue.

jeromecoutant, 2018-06-05:
I think proposed update is correct

0xc0170, 2018-06-05:
> I think proposed update is correct

The target does not yet supports bootloader, so this is not an issue (not related to the first report). I am closing this due to inactivity and if there is a request for a bootloader for a target, should be created a new ticket for this.

@martinichka you can add bootloader support for the device with the proposed change above.

closed 2018-06-06 by 0xc0170

bcostm, 2018-06-06:
See also PR #5821 (bootloader implementation for STM32F7) to see what has to be done to enable the bootloader.