ARMmbed/mbed-os #1670: I2C byte write return value

kgills, 2016-04-14:
The return value for the i2c_byte_write in i2c_api.h does not match the documentation for the write function in I2C.h.

0xc0170, 2016-04-14:
Can you please be more specific, I'll have a look

kgills, 2016-04-14:
/** Write single byte out on the I2C bus
     *
     *  @param data data to write out on bus
     *
     *  @returns
     *    '1' if an ACK was received,
     *    '0' otherwise
     */
    int write(int data);

/** Write one byte.
 *  @param obj The i2c object
 *  @param data Byte to be written
 *  @return 1 if NAK was received, 0 if ACK was received, 2 for timeout.
 */
int  i2c_byte_write(i2c_t *obj, int data);

kgills, 2016-04-21:
Any update on this?

kgills, 2016-04-26:
Please say something. I think it's pretty obvious that there's something wrong here.

0xc0170, 2016-04-26:
The problem comes:

```
int I2C::write(int data) {
    return i2c_byte_write(&_i2c, data);
}
```

Looking at HAL implementations, the documentation for the i2c_byte_write is wrong and should be fixed. The documentation to HAL was added later, so many implementation should follow what write() method defines. There are some implementations (KL25Z) which has it reversed, we should fix it. Thanks for reporting.

0xc0170, 2016-04-27:
Here's summary from our targets implementations:

Silabs - ack = 1
Freescale - ack = 0
Atmel  - ack = 0
STM - can't guess from the code, seems like timeout is 0 (might mean its nak), ack = 1 ?
NXP - ack = 1

I got a fix for the documentation, and we might need to fix all those targets.

PR #1694 (2016-04-29): i2c - fix return value description i2c_byte_write()
Fixes #1670. This follows the C++ i2c int write(int data) method, plus the implementations for nxp 1768 has it as it is with this correction, thus we can assume that's the intentional behavior. The documentation was added the last year, and was mapping some of the targets (I personally haven't noticed this).

We will add tests to get this right , we will need to fix the implementations.

@sg- @kgills
commit 884e1ade (2016-04-29): i2c - fix return value description i2c_byte_write() (#1694)

Fixes #1670.
closed 2016-04-29 by 0xc0170