ARMmbed/mbed-os #1671: Exporters and toolchains - flags to be shared

0xc0170, 2016-04-15:
Toolchain scripts have defined flags internal in the methods (mostly __init__). Therefore exporters ducplicate some but not all.

As an example for uvision (ARM) the flash size reported in the online IDE  for BLE Thermometer (mkit target) is around 28k flash, exported to uvision - 48k flash.

For ARM class, in the ``__init__``:
```
        main_cc = join(ARM_BIN, "armcc")
        common = ["-c",
            "--cpu=%s" % cpu, "--gnu",
            "-Otime", "--split_sections", "--apcs=interwork",
            "--brief_diagnostics", "--restrict", "--multibyte_chars"
        ]
```

In the uvision template, --restrict --multibyte_chars -Otime --split_sections flags are not set. This should be unified. 

I imagine a toolchain class have methods:  ``get_common_flags``, ``get_c_flags``, ``get_ccp_flags``, and ``get_ld_flags``.

theotherjimmy, 2016-06-07:
I'll be using properties instead of getters, but the idea is the same. http://stackoverflow.com/questions/6618002/python-property-versus-getters-and-setters

theotherjimmy, 2016-06-28:
#1911 is merged so this is done then? @0xc0170

theotherjimmy, 2016-09-09:
This has been done a long time ago, lets get this closed out @sg-

closed 2016-09-09 by sg-