ARMmbed/mbed-os #1672: Nucleo F103RB ADC behaves different from other Nucleo boards using MBED lib

peeyushpashine, 2016-04-15:
In case of Nucleo F103RB why the first time ADC timeouts as image below and return zero, while 2nd time onwards it catches the EOC flag. Does external trigger for ADC takes time to raise TIM1_CC1 in order to set the EOC and complete the ADC conversion? What is the reason behind this behavior which does not occur with other Nucleo boards?

[image]

In Nucleo F401RE board:
AdcHandle.Init.ExternalTrigConvEdge  = ADC_EXTERNALTRIGCONVEDGE_NONE;
while this is not set in F103RB board, which makes external trigger responsible for ADC trigger. 

[image]

However in F401RE ADC is triggered by software trigger. 

[image]

Why the behavior for F103RB is intended to use external trigger? How often does it get trigger based on TIM1_CC1? What is I want to set sampling rate as different, would timer allow me here. If ADC clock is 12.5 mhz, I see for 16 bit counter the conversion rate should come around 5ms. Is this rate sufficient enough?

0xc0170, 2016-04-15:
cc @bcostm @adustm

peeyushpashine, 2016-04-20:
I am sorry but this issue is blocking for me. I am shipping Simulink blocks for ST Nucleo hardware based on MBED libs. Can anyone give me a reason behind the different behavior mentioned in the issue?

bcostm, 2016-04-20:
Hi,
The code you are mentionning in the ST HAL driver. We know that the F1 HAL driver version is not the latest version in mbed. We plan to update it in a couple of weeks. But this doesn't mean that this behaviour will be changed. What you can do in the mean time, if possible for you, is to download and use the latest HAL driver in your mbed fork. Or did you look into ST forum www.st.com/e2e ?

peeyushpashine, 2016-04-21:
Yes I looked into  www.st.com/e2e but did not find any threads on this and was not answered for my post yet. Is there a reason behind the different behavior for ST HAL driver for Nucleo F103rb board? Why does it uses external trigger rather than software trigger as the other hardware boards.

bcostm, 2016-05-11:
Hi,
Can you try this in the analogin_api.c file ?
Replace:
    AdcHandle.Init.ExternalTrigConv      = ADC_EXTERNALTRIGCONV_T1_CC1;
By:
    AdcHandle.Init.ExternalTrigConv      = ADC_SOFTWARE_START;

peeyushpashine, 2016-05-17:
Thanks for the update. That is what I have done for the hack as of now. Do you have any plans to update this?

bcostm, 2016-05-17:
Hi,
I can't tell you when we'll update it. But everybody is encouraged to participate in the SDK development,  not only ST :smiley: 
If you have already done it and tested it OK, you can send a Pull Request. It will be merged quicky I think.

bcostm, 2016-09-09:
Can you please close this issue if the problem is solved ?

closed 2016-09-18 by peeyushpashine