ARMmbed/mbed-os #1683: FRDM-K64F export to GCC doesn't work

janjongboom, 2016-04-22:
Just says 'Downloading...' but never actually downloads anything.

mlnx, 2016-04-22:
@screamerbg

0xc0170, 2016-05-05:
I think there should be issue tracker for the web tools. I assumed always it's under bug & suggestions on the developer.mbed.org.. This should not be a tracker for them

adamgreen, 2016-05-05:
By the way, I hit something similar last night.  It said downloading for a minute or so and then erred out with a message about how it couldn't do the export for my selected target. I started doing something else and then it completed in the background and I had the .zip that I wanted.

sg-, 2016-05-06:
@adamgreen Was this using the mbed library in source or binary??

adamgreen, 2016-05-06:
@sg- It should have been binary. I had just used the "Create new program..." dialog and selected the default frdm_gpio template. I immediately tried to export it to GCC_ARM right after the new project was created (I wanted to look at the Makefile) and hit that issue. 

I was able to do the exact same steps today with no issues.

infinnovation, 2016-08-18:
This may or may not be related to an issue with export to GCC ARM Embedded which I have [just posted](https://developer.mbed.org/forum/bugs-suggestions/topic/26579/) on the Bugs & Suggestions forum (bearing in mind @0xc0170's comment above)

0xc0170, 2016-08-18:
cc @screamerbg Please have a look at the link in the above comment

theotherjimmy, 2016-10-20:
I'm pretty sure this is resolved. It has been a while though.

closed 2016-10-20 by theotherjimmy