ARMmbed/mbed-os #1685: STM32 HAL issue - reading internal temp after VBAT measurement.

campag, 2016-04-23:
Measuring VBAT sets the ADC_CCR_VBATE bit in ADC->CCR, and there doesn't appear to be a way with the HAL to clear it.

If it isn't cleared, VBAT remains connected to the ADC channel in preference to temperature, so VBAT readings are returned in place of temperature. I resorted to poking the register directly:
```
float read_temp(void)
{
    ADC_ChannelConfTypeDef sConfig = {0};

    // Configure ADC channel
    sConfig.Rank         = 1;
    sConfig.SamplingTime = ADC_SAMPLETIME_84CYCLES;
    sConfig.Offset       = 0;
    sConfig.Channel      = ADC_CHANNEL_TEMPSENSOR;

    HAL_ADC_ConfigChannel(&hAdc, &sConfig);
    
    //disable Vbat from  ADC1_IN18 input channel, shared with temperature sensor
    ADC->CCR &= ~ADC_CCR_VBATE;
    //better added instead in stm32f4xx_hal_adc.c at line 1123, also line 753 in adc_ex.c?
    
    HAL_ADC_Start(&hAdc); // Start conversion

    // Wait end of conversion and get value
    if (HAL_ADC_PollForConversion(&hAdc, 10) == HAL_OK) {
        uint16_t value=HAL_ADC_GetValue(&hAdc);
        HAL_ADC_Stop(&hAdc);
        return  25.0f+(3.3f*(float)value/(float)ADCMAXVALUE-0.76f)/0.0025f;
    } else {
        return -273.15f; //hell has frozen over
    }
}
```

As per the comments in above, perhaps that bit should be cleared when configuring the ADC to read the temperature in stm32f4xx_hal_adc.c & stm32f4xx_hal_adc_ex.c ?

bcostm, 2016-04-25:
Hi, thanks for pointing that out. We'll have a deeper look when we'll add the ADC internal channels measurement in mbed Classic. FYI we have started doing this in mbed OS on another device (F0, but not public for the moment). I'll check if there is the same problem.

campag, 2016-04-26:
Hi. I have seen such ADC-internal extensions to mbed(os?) for the F4 over at https://github.com/ARMmbed which might benefit from clearing that bit too.

bcostm, 2016-11-18:
A patch is on-going in mbed SDK (not ST HAL driver).

closed 2017-01-03 by campag