ARMmbed/mbed-os #1686: Keil uvision exporter fails (uARM build target)

toyowata, 2016-04-25:
The uvison exporter now fails for most of uARM toolchain platforms which worked find a month ago (or two weeks ago?).

* LPC1768 - OK (ARM)

* LPC11U24 - Link error (Target is ulib, but scatter file is non-ulib)
```
.\build\mbed_blinky.axf: Error: L6218E: Undefined symbol __initial_sp (referred from entry2.o).
.\build\mbed_blinky.axf: Error: L6218E: Undefined symbol __heap_base (referred from malloc.o).
.\build\mbed_blinky.axf: Error: L6218E: Undefined symbol __heap_limit (referred from malloc.o).
```
* KL25Z - OK (ARM)

* LPC812-MAX - Link error (no mbed.ar and other .o files)
```
.\build\mbed_blinky.axf: Error: L6320W: Ignoring --entry command. Cannot find argument 'Reset_Handler'.
```

* Seeed Arch - Link error (Target is ulib, but scatter file is non-ulib)
```
.\build\mbed_blinky.axf: Error: L6218E: Undefined symbol __initial_sp (referred from entry2.o).
.\build\mbed_blinky.axf: Error: L6218E: Undefined symbol __heap_base (referred from malloc.o).
.\build\mbed_blinky.axf: Error: L6218E: Undefined symbol __heap_limit (referred from malloc.o).
```

* LPC1114FN28 - Link error (no mbed.ar and other .o files)
```
.\build\mbed_blinky.axf: Error: L6320W: Ignoring --entry command. Cannot find argument 'Reset_Handler'.
```

This is reported here: https://developer.mbed.org/questions/68792/LPC1114FN28uVision/

toyowata, 2016-04-25:
How to reproduce:
* Create a blinky program
* Set uARM target (such as LPC1114FN28)
* Export to uvision (Keil uVision4)

0xc0170, 2016-04-25:
Thanks for reporting. I'll fix it within 2 days, seems like ulib template is at fault

PR #1687 (2016-04-26): uvision - get_toolchain method regression fix
uvision uses default_toolchain, thus exporter needs to provide a method to
distinguish between ARM and uARM. Fixes #1686 

@toyowata Please test. I tested this with KL05 (uARM) and KL25Z (ARM) if proper toolchain is set
commit 864b3494 (2016-04-26): uvision - get_toolchain method regression fix

uvision uses default_toolchain, thus exporter needs to provide a method to
distinguish between ARM and uARM. Fixes #1686
closed 2016-04-26 by 0xc0170