ARMmbed/mbed-os #1689: bug in analogout_write_u16() for Arch MAX

manitou48, 2016-04-26:
This is the same problem I reported in issue #1383 now i'm seeing it on Arch MAX.  Documentation says dac.write_u16() value is 16 bits (and so it is on LPC1768, K64F, nucleo_F446RE), but its only 12 bits on Arch MAX.

0xc0170, 2016-04-26:
Does not Arch max use the same HAL as F446RE? It's STM32F4 chip.

cc @xiongyihui

manitou48, 2016-04-26:
On Tue, Apr 26, 2016 at 11:13:46AM -0700, Martin Kojtal wrote:
> Does not Arch max use the same HAL as F446RE? It's STM32F4 chip.
> 
> cc @xiongyihui

I would think so.  And I looked at the core code and didn't really
see a problem.  I re-ran my tests on LPC, K64F, and F446RE, and they all
are using 16-bit.  i'll re-run on Arch MAX.  (I can't see that it should
matter, but i'm using Arch MAX to generate .bin for STM32F4-DISCOVERY
board.)

manitou48, 2016-04-26:
Well, I have copies of the output from yesterday, but today I cannot reproduce the problem.  DAC is writing in 16-bit mode as it should.     Whatever.  should close and ignore.

closed 2016-04-26 by manitou48