ARMmbed/mbed-os #1692: progen template for nrf51 targets

BlackstoneEngineering, 2016-04-26:
## Problem:
New Progen export system is not fully functional for nrf51 targets. Currently when exported (Ex Keil) the hex being built contains just the program for the user code. Previously the generated hex would contain both the nordic soft device and the user space code. See [here](https://github.com/mbedmicro/mbed/blob/604e163cc1c875f263744b0db63f1d1593cfd323/workspace_tools/export/uvision4_arch_ble.uvproj.tmpl#L79)

## Solution:
Add soft device to exported zip file and add post processing command to build to merge soft device and user code. Might be necessary to include mergehex or similar tool in the exported zip file to ensure success.

@0xc0170 @rgrover

theotherjimmy, 2016-10-20:
Progen is no longer used.

sg-, 2017-01-16:
Most IDEs use an elf to load the application binary into the target flash. Softdevice is delivered as a hex file so this would mean that we'd need a post build step to generate an application hex, merge it with softdevice and then configure the IDE to load the hex, not the elf.

While its possible to create a application hex file that included both softdevice and application, I'm not certain it was loaded if using CMSIS-DAP when programming in the IDE although it could be dragged and dropped onto the board.

Should post build steps done in mbed tools be applied to all IDEs when exported?

BlackstoneEngineering, 2017-01-17:
I would argue it should, otherwise you have an inconsistent user experience. (works in the online IDE, doesnt in the offline IDE)

theotherjimmy, 2017-01-17:
I argue that the architecture prevents it, in general. ATM, we default to merging when there is a hex file present, when exporting to make. I think that covers many of the post build scripts.

BlackstoneEngineering, 2017-01-17:
This will be critical for modules going forward, being able to link together the black box binary drop from the manufacturer and our code will be critical. So having some way to do this across all toolchains is worthwhile in my opinion.

bridadan, 2017-01-17:
@BlackstoneEngineering I think what @theotherjimmy means is that certain IDEs don't let you add a post build script. OR if they do they don't allow us to do all the operations we need to do.

theotherjimmy, 2017-01-17:
@bridadan Yes, and we have no way of exporting a generic post-build script, as they are all baked into tools/targets.py.

sg-, 2017-01-17:
Need to have a good hard think about this if bootloader and other combinatorial applications are going to be possible. Not supporting exporters isn't an acceptable answer 👅

BlackstoneEngineering, 2017-01-18:
I thought this already worked for the NRF targets from the online compiler? I know it works in Keil for sure, not sure how / if it was implemented in IAR

theotherjimmy, 2017-01-18:
> but the idea of running a script after the build should be possible with all IDE's.

This is not the problem.

The problem is that the current implementation of the post-build step is to have a python *function*. This cannot export, as it is only one part of a file (`tools/targets.py` to be exact), and that file imports, well, the rest of the tools.

bridadan, 2017-01-18:
@theotherjimmy so to support this we need to establish that post-build steps need to be stand-alone scripts with no dependencies on the mbed tools (but potentially external pip packages that would be in mbed-os's requirements.txt). Does that sound right? That sounds like that should be achievable. All data that needs to be passed to the script should be done via the command line or via function arguments.

theotherjimmy, 2017-01-18:
>Does that sound right? 

Yes

bridadan, 2017-01-18:
@theotherjimmy Ok cool, sounds like the best way to move forward on resolving this issue (probably should close this and create a new one that's more relevant) is to submit a PR that refactors all existing post-build steps in this way, and document how they should be written/organized. Not saying you need to do this, just double checking that this is a good plan moving forward.

theotherjimmy, 2017-01-18:
@bridadan 👍 💯

BlackstoneEngineering, 2017-01-18:
I would say rather than add more pip packages to simply ship some of the simpler tools inside of the mbed tools directory. As @sg- once said `we can write mergehex in ~100 lines of scripting and just ship it with every export`. 

My thought is instead of installing more things to the system to instead have standalone scripts that ship with our tools and thus provide a certain number of possible scripts. The most important and immediate being something like mergehex. Maybe as we go forward other tools will become important as well, but that is very hard to predict.

bridadan, 2017-01-18:
@BlackstoneEngineering We already install the intelhex package with the tools that takes care of this for us. If they export with mbed CLI they will get it automatically. But I agree we (and vendors) shouldn't add pip packages willy-nilly.

closed 2017-10-27 by ChiefBureaucraticOfficer