ARMmbed/mbed-os #1701: Instable ticker on stm32f0308-DISCO

joost-g, 2016-05-02:
Hi,

I ran into an issue using mbed's ticker on the stm32f0308-DISCO platform.
As can be seen from the scope images, the ticker callback is not called for a period, after which it is called multiple times. Issue occurs approximately twice every second.

Minimal working example to reproduce:
```
#include "mbed.h"

Ticker msTicker;
DigitalOut dev_led4(PC_8);

static void ticker_callback(){
	dev_led4=1;
	volatile int i;
	for(i=0; i<2500;i++);
	dev_led4=0;
}

int main(void){
	dev_led4 = 0;

    msTicker.attach(&ticker_callback, 0.01);
	while(1);
}
```
Normal/expected behavior
[image]
Non-uniform ticker callback
[image]
[image]

0xc0170, 2016-05-02:
@bcostm @adustm

JojoS62, 2016-05-02:
This problem looks like the same issue on the STM32F103 : https://github.com/mbedmicro/mbed/issues/816
The code is the same for the F1 and the problem is the extension of the 16 bit counter/timer, it is not working as intended. Maybe a better solution is to concat two 16 bit timers, the ST timers have this option but I haven't tried it yet.

bcostm, 2016-05-02:
Same issue as #816 on the 16-bit timer. The problem is understood and is under validation.

Dieter-Graef, 2016-05-06:
I had a same issue on the STM32F746 Discovery board. The reason was that the ticker requires a free running counter to work. Therefor the autoupdate feature  at reaching zero needs to be switched off. I added therefor the following line in the function HAL_InitTick:
    TimMasterHandle.Instance = TIM_MST;
    TimMasterHandle.Init.Period = 0xFFFFFFFF;

    // TIMxCLK = PCLKx when the APB prescaler = 1 else TIMxCLK = 2 * PCLKx
    if (RCC_ClkInitStruct.APB1CLKDivider == RCC_HCLK_DIV1)
      TimMasterHandle.Init.Prescaler = (uint16_t)((PclkFreq) / 1000000) - 1; // 1 us tick
    else
      TimMasterHandle.Init.Prescaler = (uint16_t)((PclkFreq * 2) / 1000000) - 1; // 1 us tick

    TimMasterHandle.Init.ClockDivision     = 0;
    TimMasterHandle.Init.CounterMode       = TIM_COUNTERMODE_UP;
    TimMasterHandle.Init.RepetitionCounter = 0;
    HAL_TIM_OC_Init(&TimMasterHandle);
    base[0].CR1|=TIM_CR1_UDIS;  //<- I added this
    NVIC_SetVector(TIM_MST_IRQ, (uint32_t)timer_irq_handler);
    NVIC_EnableIRQ(TIM_MST_IRQ);

sg-, 2017-01-16:
@joost-g @bcostm is this resolved?

LMESTM, 2017-06-01:
@sg- @joost-g @bcostm 
There is a fix for this issue here - autoamtic tests are running now
anyone willing to do more tests is welcome
https://github.com/LMESTM/mbed/tree/STM32_16bits_tickers

LMESTM, 2017-06-09:
ST_TO_BE_CLOSED
#4424 was merged - @joost-g  the problem is fixed now, feel free to let us know if any further issue

closed 2017-06-09 by 0xc0170