ARMmbed/mbed-os #1708: Can't anymore test ARM and uARM toolchain

svastm, 2016-05-04:
After the merge #1691 it's not anymore possible to use the command singletest.py with both uARM and ARM toolchain. The ARM part fail build when uARM is also used. If ARM is used alone, no problem.

The problem come from `arm.py`. The ARM_STD and ARM_MICRO class are sharing a reference `self.flags` to `DEFAULT_FLAGS` and when ARM_MICRO object is created, it add, amongst others, `--library_type=microlib` in the `DEFAULT_FLAG` and then in the `self.flags` of ARM_STD object.

A fix could be to copy the `DEFAULT_FLAGS` line 57.

```python
self.flags = copy.deepcopy(self.DEFAULT_FLAGS)
```

Another problem is that the uARM toolchain is always in timeout. Can you verify this point ?

bridadan, 2016-05-04:
Hi @svastm, I'm having trouble reproducing your issue on the K64F. Are you using the latest master? If so and you're still seeing the same issue, could please post the following info?

1. The platform you're testing
2. The commands you're using
3. The errors you're seeing
4. The version of armcc you're using

svastm, 2016-05-04:
I have the issue with, at least, NUCLEO_L031K6. and the updated master (HEAD: `9cef243de23875778f461bbe9a8c1bc47e65212b`) when I use the following command:
```
python ..\mbed\workspace_tools\singletest.py --auto -f NUCLEO_L031K6,NUCLEO_L053R8,NUCLEO_L073RZ,DISCO_L053C8 --parallel --tc uARM,ARM,IAR,GCC_ARM --global-loops=1 --clean -j 4 -c copy -V -v --report-html=.\reports\report__2016-05-04_12.59.53.67__L0__common__ALL.html -C
```

with armcc version 5.6.0.13.

During an ARM_STD compilation, I have the following line where the options `--library_type=microlib` and `-D__MICROLIB` should not be present.

```
Compile: main.cpp
[DEBUG] Command: C:\Keil\ARM\ARMCC\bin\armcc -c --gnu -Otime --split_sections --pcs=interwork --brief_diagnostics --restrict --multibyte_chars --cpu=Cortex-M0 -O3 --cpu=Cortex-M0 -O3 --cpu=Cortex-M0 -O3 --cpu=Cortex-M0 -O3 --md --no_depend_system_headers -IC:\Keil\ARM\ARMCC\include --c99 -D__ASSERT_MSG --library_type=microlib -D__MICROLIB --library_type=microlib -D__MICROLIB --cpp --no_rtti -D__ASSERT_MSG --library_type=microlib -D__MICROLIB --library_type=microlib -D__MICROLIB -D__CORTEX_M0 -DTEST_SUITE_UUID="67e26a5a-17ca-40cf-8ca4-988dac97eda9" -DTARGET_FF_ARDUINO -DTEST_SUITE_TARGET_NAME="NUCLEO_L031K6" -DTARGET_CORTEX_M -DTARGET_LIKE_MBED -DARM_MATH_CM0 -DTOOLCHAIN_ARM_STD -DMBED_BUILD_TIMESTAMP=1462360969.32 -DTARGET_STM32L0 -DTARGET_STM -D__MBED__=1 -DTARGET_M0 -DTARGET_STM32L031K6 -DTEST_SUITE_TEST_ID="MBED_37" -DTARGET_NUCLEO_L031K6 -DTOOLCHAIN_ARM -DTARGET_LIKE_CORTEX_M0 -IC:\Users\xxx\github\mbed\libraries\tests\mbed\serial_nc_rx -IC:\Users\xxx\github\mbed\libraries\tests\mbed\serial_nc_rx -IC:\Users\xxx\github\mbed\build\mbed -IC:\Users\xxx\github\mbed\build\mbed\TARGET_NUCLEO_L031K6 -IC:\Users\xxx\github\mbed\build\mbed\TARGET_NUCLEO_L031K6\TARGET_STM -IC:\Users\xxx\github\mbed\build\mbed\TARGET_NUCLEO_L031K6\TARGET_STM\TARGET_STM32L0 -IC:\Users\xxx\github\mbed\build\mbed\TARGET_NUCLEO_L031K6\TARGET_STM\TARGET_STM32L0\TARGET_NUCLEO_L031K6 -IC:\Users\xxx\github\mbed\build\mbed\TARGET_NUCLEO_L031K6\TOOLCHAIN_ARM_STD -IC:\Users\xxx\github\mbed\libraries\tests\mbed\env -o C:\Users\vansons\github\mbed\build\test\NUCLEO_L031K6\ARM\MBED_37\.\main.o C:\Users\xxx\github\mbed\libraries\tests\mbed\serial_nc_rx\main.cpp --depend C:\Users\xxx\github\mbed\build\test\NUCLEO_L031K6\ARM\MBED_37\.\main.d
```
Then the link fail : 
```
Link: serial_nc_rx
Error: L6218E: Undefined symbol __initial_sp (referred from entry2.o).
Error: L6218E: Undefined symbol __heap_base (referred from malloc.o).
Error: L6218E: Undefined symbol __heap_limit (referred from malloc.o).
Finished: 0 information, 0 warning and 3 error messages.
```

0xc0170, 2016-05-04:
@svastm Thanks for reporting, I'll have a look at those microlib flags

0xc0170, 2016-05-04:
Should be fixed. I was able to run uARM test, and also both ARM and uARM. Please verify, and reopen with more info if this problem persists.

Thanks for the description with the solution for this bug, appreciated :+1:

closed 2016-05-04 by 0xc0170