ARMmbed/mbed-os #1721: STM-L476 inceased sleep power problem

star297, 2016-05-07:
@bcostm 
Since one the recent updates for the L476 target, the deepsleep current has gone way off spec.
We now have around 115uA instead of 5 - 6uA.
Working on revision 53 but not sure which revision this happened.
Could you take a look please.

bcostm, 2016-09-09:
Can you please close this issue if the consumption problem is solved ?

In PR #1832
> Current consumption is correct on both LSI and LSE modes with RTC functioning as expected.

sg-, 2017-01-16:
@star297 Is this resolved?

helmut64, 2017-01-16:
The problem was resolved until rev#132 it worked fine, it is again broken in #133 (issue #3542). It has been fixed according to my input (HAL_PWREx_EnableLowPowerRunMode). I verified the head version it it works now.
The case can be closed.

closed 2017-01-16 by 0xc0170