ARMmbed/mbed-os #1723: Exporters - uvision - LPC11U68

chrissnow, 2016-05-09:
The definition is incorrect, it looks like a copy of another 11u device

the correct info wold be
```
<Cpu>IRAM(0x10000000-0x10007FFF) IRAM2(0x20000000-0x200007FF) IROM(0x0-0x3FFFF) CLOCK(12000000) CPUTYPE("Cortex-M0+")</Cpu>

<FlashDriverDll>UL2CM3(-O4303 -S0 -C0 -FO7 -FD10000000 -FC800 -FN1 -FF0LPC1xxx_96_160 -FS00 -FL040000)</FlashDriverDll>

<DeviceId>7280</DeviceId>

<SFDFile>SFD\NXP\LPC11Uxx\LPC11U6x.SFR</SFDFile>
```

0xc0170, 2016-05-09:
@chrissnow Please report this to https://github.com/project-generator/project_generator_definitions/ . The module (you probably have it installed along with progen) contains a parser for project files. You should be able to get the proper yaml file, and add it to teh current one , send a patch.

0xc0170, 2016-05-09:
https://github.com/project-generator/project_generator_definitions/blob/master/project_generator_definitions/mcu/nxp/lpc11u68.yaml , this is the file correct? Does it return an error when opening a project or what is the failure? I could compare the old templates, if they were not parsed correctly or they were wrong.

chrissnow, 2016-05-09:
Hi,

That's the correct file though the contents are all wrong, it looks like it was copied from the LPC11U14/201 rather than the 11u68

I will look into fixing it and send it over.

chrissnow, 2016-05-09:
this should have the correct info in for reference.
[uvproj.txt](https://github.com/mbedmicro/mbed/files/255416/uvproj.txt)

0xc0170, 2016-05-09:
@chrissnow  Please send a patch. run ``progendef create`` (use help to get arguments)

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-