ARMmbed/mbed-os #1726: RZ_1AH - IAR does not build

0xc0170, 2016-05-10:
The error indicates a problem with cmsis - ``__disable_irq`` is used in the serial_api.c which is not compatible with IAR - there was added __disable_irq_iar().  It should probably not have iar prefix? Plus the cache functions are neither implemented for IAR.

@TomoYamanaka

TomoYamanaka, 2016-05-11:
We have confirmed the phenomenon, and analyzed the cause.
In the following revision, we found that the phenomenon is different.
&nbsp; &nbsp;mbed Lib revision.119 : Build NG
&nbsp; &nbsp;mbed Lib revision.118 : Build OK
&nbsp; &nbsp;mbed-dev revision.125 : Build OK

When you update to mbed Lib revision.119, it seems that the code for IAR has gone missing.

We have already corresponed the export function of IAR in the following PR.
https://github.com/mbedmicro/mbed/pull/1545
https://github.com/mbedmicro/mbed/pull/1590

toyowata, 2016-05-17:
@TomoYamanaka 

```
[Error] serial_api.c@527: [Pe513]: a value of type "void" cannot be assigned to
an entity of type "int"

          was_masked = __disable_irq();
                     ^
```
The #1626 seems to be a cause of error for IAR tool.

https://github.com/mbedmicro/mbed/commit/418fd7ec94c89e9eb5da5db829296d59fa25179d#diff-2467556b51fd19c28f6bb9c264dddd68R529

It should have ```#if defined ( __ICCARM__ )``` definition as well.
Can you create new PR to fix this?

TomoYamanaka, 2016-05-17:
@toyowata 

Sorry. 
There was an error in the comment.
In mbed-dev revision.125, IAR did not build.
As you pointed out, there was an error in Serial_api.c (Line 527). 
I'll create new PR to fix this.

However, the following behavior occurs.
&nbsp; &nbsp; mbed Lib revision.119 : Build NG
&nbsp; &nbsp; mbed Lib revision.118 : Build OK
When you update to mbed Lib revision.119, it seems that the code for IAR has gone missing.
core_caFunc.c files, etc.

0xc0170, 2016-05-17:
> When you update to mbed Lib revision.119, it seems that the code for IAR has gone missing.
core_caFunc.c files, etc.

RZ_A1H IAR had to be disabled as it did not build, we should fix it for the upcoming release.

toyowata, 2016-09-23:
@0xc0170 I confirmed that exported IAR project for RZ/A1H works fine.  Ready for close this issue?

closed 2016-09-23 by 0xc0170