ARMmbed/mbed-os #1736: STM32F7 : rtos tests don't build with IAR

adustm, 2016-05-12:
Build fail with IAR toolchain for rtos tests

[report__2016-05-11_14.45.44.48__NUCLEO-F746ZG__rtos__ALL.html.zip](https://github.com/mbedmicro/mbed/files/261476/report__2016-05-11_14.45.44.48__NUCLEO-F746ZG__rtos__ALL.html.zip)

adustm, 2016-05-12:
HAL_CM4.S cannot compile :

```
$ ../../../../Program\ Files\ \(x86\)/IAR\ Systems/Embedded\ Workbench\ 7.3/arm
/bin/iasmarm.exe --cpu Cortex-M7 -DTARGET_STM32F746ZG -D__CORTEX
_M7 -DTARGET_STM32F746 -DARM_MATH_CM7 -DTARGET_CORTEX_M -D__FPU_PRESENT=1 -DTAR
GET_LIKE_CORTEX_M7 -D__MBED__=1 -DTARGET_RTOS_M4_M7 -DTARGET_STM -DTARGET_M7 -D
TARGET_STM32F7 -DTOOLCHAIN_IAR -DMBED_BUILD_TIMESTAMP=1463067288.13 -DTARGET_NU
CLEO_F746ZG -DTARGET_LIKE_MBED -I../../libraries/rtos/rtx -I../../libraries/rto
s/rtx/TARGET_CORTEX_M -I../../libraries/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4
_M7 -I../../libraries/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_IAR
-I../../build/mbed -I../../build/mbed/TARGET_NUCLEO_F746ZG -I../../build/mbed/T
ARGET_NUCLEO_F746ZG/TARGET_STM -I../../build/mbed/TARGET_NUCLEO_F746ZG/TARGET_S
TM/TARGET_STM32F7 -I../../build/mbed/TARGET_NUCLEO_F746ZG/TARGET_STM/TARGET_STM
32F7/TARGET_NUCLEO_F746ZG -I../../build/mbed/TARGET_NUCLEO_F746ZG/TOOLCHAIN_IAR
 -I../../build/rtos -I../../build/rtos/TARGET_CORTEX_M -I../../build/rtos/TARGE
T_NUCLEO_F746ZG -I../../build/rtos/TARGET_NUCLEO_F746ZG/TOOLCHAIN_IAR -o ./HAL_
CM4.o ../../libraries/rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_IAR/
HAL_CM4.S
IAR Assembler V7.50.3.10732/W32 for ARM
Copyright 1999-2016 IAR Systems AB.

        VSTMDBEQ R12!,{S16-S31}         /* yes, stack also VFP hi-regs */
--------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",192  Error[438]: This instruction is not available in the selected cpu/core

        VLDMIANE R12!,{S16-S31}         /* restore VFP hi-registers */
--------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",213  Error[438]: This instruction is not available in the selected cpu/core

        VSTMDBEQ R12!,{S16-S31}         /* yes, stack also VFP hi-regs */
--------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",278  Error[438]: This instruction is not available in the selected cpu/core

        VLDMIANE R12!,{S16-S31}         /* restore VFP hi-registers */
--------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",298  Error[438]: This instruction is not available in the selected cpu/core

        VSTMDBEQ R12!,{S16-S31}         /* yes, stack also VFP hi-regs */
---------------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",192  Error[50]: Undefined symbol:'S16'
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",192  Error[50]: Undefined symbol:'S31'

        LSLS    R0,#31
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",92  Warning[408]: Not recommended operand syntax. Duplicate the first register operand, or select option -j (alternative syntax).

        LSLS    R3,#24
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",110  Warning[408]: Not recommended operand syntax. Duplicate the first register operand, or select option -j (alternative syntax).

        LSLS    R3,#31
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",114  Warning[408]: Not recommended operand syntax. Duplicate the first register operand, or select option -j (alternative syntax).

        LSLS    R3,#24
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",131  Warning[408]: Not recommended operand syntax. Duplicate the first register operand, or select option -j (alternative syntax).

        LSLS    R3,#31
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_IAR\HAL_CM4.S",135  Warning[408]: Not recommended operand syntax. Duplicate the first register operand, or select option -j (alternative syntax).

Errors:   6
Warnings: 5
```

The VFPv5_sp option is missing for assembly code
```
$ ../../../../Program\ Files\ \(x86\)/IAR\ Systems/Embedded\ Workbench\ 7.3/arm
[... 37 lines cut ...]
        LSLS    R3,#24
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHA
IN_IAR\HAL_CM4.S",131  Warning[408]: Not recommended operand syntax. Duplicate t
he first register operand, or select option -j (alternative syntax).

        LSLS    R3,#31
----------------^
"C:\MCDGitW601\mbed\libraries\rtos\rtx\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHA
IN_IAR\HAL_CM4.S",135  Warning[408]: Not recommended operand syntax. Duplicate t
he first register operand, or select option -j (alternative syntax).

Errors:   0
Warnings: 5
Bytes: 404
```

iar.py needs to be modified to add this option. I tried
```
63        if target.core == "Cortex-M7F":
64           self.asm  = [join(IAR_BIN, "iasmarm")] + ["--cpu", cpuchoice] + ["--fpu VFPv5_sp"]
65        else:
66           self.asm  = [join(IAR_BIN, "iasmarm")] + ["--cpu", cpuchoice]
```
but it does not work. Any other suggestion ?

0xc0170, 2016-05-12:
Thanks for reporting, I'll look into this week. There was not much update in those files, so worth comparing to the older version.

adustm, 2016-05-13:
>  so worth comparing to the older version.
I think it is the first time that we compile rtos for cortex M7.
I had already made a modification in the past [https://github.com/mbedmicro/mbed/pull/1604](url) to allow compiling mbed for STM32F7 with IAR toolchain. 
It seems that this previous modification was not strong enough for rtos library.

adustm, 2016-05-13:
here is the build command that fails with the attached iar.py file (rename it to .py)
`python ..\mbed\workspace_tools\build.py --rtos -v -x -c -t IAR -m NUCLEO_F746ZG -j 4`

It reports : 
```
Building library CMSIS (NUCLEO_F746ZG, IAR)
Compile: startup_stm32f746xx.S
[DEBUG] Command: C:\Program Files (x86)\IAR Systems\Embedded Workbench 7.3\arm\bin\iasmarm --cpu Cortex-M7 --fpu VFPv5_s
p  -DMBED_BUILD_TIMESTAMP=1463128750.73 -DTARGET_STM32F746ZG -D__CORTEX_M7 -DTARGET_STM32F746 -DARM_MATH_CM7 -DTARGET_CO
RTEX_M -D__FPU_PRESENT=1 -DTARGET_LIKE_CORTEX_M7 -D__MBED__=1 -DTARGET_RTOS_M4_M7 -DTARGET_STM -DTARGET_M7 -DTARGET_STM3
2F7 -DTOOLCHAIN_IAR -DTARGET_NUCLEO_F746ZG -DTARGET_LIKE_MBED -IC:\MCDGitW601\mbed\libraries\mbed\targets\cmsis -IC:\MCD
GitW601\mbed\libraries\mbed\targets\cmsis\TARGET_STM -IC:\MCDGitW601\mbed\libraries\mbed\targets\cmsis\TARGET_STM\TARGET
_STM32F7 -IC:\MCDGitW601\mbed\libraries\mbed\targets\cmsis\TARGET_STM\TARGET_STM32F7\TARGET_NUCLEO_F746ZG -IC:\MCDGitW60
1\mbed\libraries\mbed\targets\cmsis\TARGET_STM\TARGET_STM32F7\TARGET_NUCLEO_F746ZG\TOOLCHAIN_IAR -IC:\MCDGitW601\mbed\li
braries\mbed\targets\cmsis\TOOLCHAIN_IAR -o C:\MCDGitW601\mbed\build\mbed\.temp\TARGET_NUCLEO_F746ZG\TOOLCHAIN_IAR\TARGE
T_STM\TARGET_STM32F7\TARGET_NUCLEO_F746ZG\TOOLCHAIN_IAR\startup_stm32f746xx.o C:\MCDGitW601\mbed\libraries\mbed\targets\
cmsis\TARGET_STM\TARGET_STM32F7\TARGET_NUCLEO_F746ZG\TOOLCHAIN_IAR\startup_stm32f746xx.S
[DEBUG] Return: 3
Traceback (most recent call last):
  File "..\mbed\workspace_tools\build.py", line 234, in <module>
    macros=options.macros)
  File "C:\MCDGitW601\mbed\workspace_tools\build_api.py", line 450, in build_mbed_libs
    raise e
ToolException
```

[iar.py.txt](https://github.com/mbedmicro/mbed/files/262840/iar.py.txt)

adustm, 2016-05-13:
I found the issue in iar.py:
```
63        if target.core == "Cortex-M7F":
64           self.asm  = [join(IAR_BIN, "iasmarm")] + ["--cpu", cpuchoice] + ["--fpu", "VFPv5_sp"]
65        else:
66           self.asm  = [join(IAR_BIN, "iasmarm")] + ["--cpu", cpuchoice]
```
I'll push the pull request.
Cheers
Armelle

adustm, 2016-09-12:
Compilation error is fixed with current mbed-os database
`../../../Program\ Files\ \(x86\)/IAR\ Systems/Embedded\ Workbench\ 7.3/arm/bin/iasmarm.exe --cpu Cortex-M7 --fpu VFPv5_sp -DTARGET_STM32F746ZG -D__CORTEX_M7 -DTARGET_STM32F746 -DARM_MATH_CM7 -DTARGET_CORTEX_M -D__FPU_PRESENT=1 -DTARGET_LIKE_CORTEX_M7 -D__MBED__=1 -DTARGET_RTOS_M4_M7 -DTARGET_STM -DTARGET_M7 -DTARGET_STM32F7 -DTOOLCHAIN_IAR -DMBED_BUILD_TIMESTAMP=1463067288.13 -DTARGET_NUCLEO_F746ZG -DTARGET_LIKE_MBED -I../rtos/rtx -I../rtos/rtx/TARGET_CORTEX_M -I../rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7 -I../rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_IAR -I../build/mbed -I../build/mbed/TARGET_NUCLEO_F746ZG -I../build/mbed/TARGET_NUCLEO_F746ZG/TARGET_STM -I../build/mbed/TARGET_NUCLEO_F746ZG/TARGET_STM/TARGET_STM32F7 -I../build/mbed/TARGET_NUCLEO_F746ZG/TARGET_STM/TARGET_STM32F7/TARGET_NUCLEO_F746ZG -I../build/mbed/TARGET_NUCLEO_F746ZG/TOOLCHAIN_IAR -I../build/rtos -I../build/rtos/TARGET_CORTEX_M -I../build/rtos/TARGET_NUCLEO_F746ZG -I../build/rtos/TARGET_NUCLEO_F746ZG/TOOLCHAIN_IAR -o ./HAL_CM4.o ../rtos/rtx/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_IAR/HAL_CM4.S`

gives
```
Errors:   0
Warnings: 5
Bytes: 404
```

I can close the issue. Kind regards
Armelle

closed 2016-09-12 by adustm