ARMmbed/mbed-os #1737: RTOS and InterruptIn example doesn't work anymore

jeromecoutant, 2016-05-12:
Example:
https://developer.mbed.org/teams/ST/code/Nucleo_rtos/

I am testing with Nucleo-F411RE.

Test with inline librairies:
- mbed v105 : OK
- mbed v106 and after: FAILED

Test with uVision:
- ARM_MICRO compilation : Test is FAILED
- ARM_STD compilation : Test is OK

Analysis:
A InterruptIn variable is set as global.
With micro lib, during init, constructor InterruptIn is called, interruption context is then created.
Then still during init, destructor ~InterruptIn is also called, so all context is cleared.
It should be an "compilation optimization" error as the variable is then used.

Regards,

adustm, 2016-05-12:
Hello,
Isn't it related to [https://github.com/mbedmicro/mbed/pull/1432](url) and [https://github.com/mbedmicro/mbed/pull/1624](url) ?
Still the issue that global variables are cleared between systemInit and the start of the main...

cc @dinau @bcostm 
cheers
Armelle

dinau, 2016-05-14:
Hi @jeromecoutant.
 I can't read above example by access denied.
 Would you please copy and paste the example code this thread ?

dinau

jeromecoutant, 2016-05-16:
Hi

Here is the example code:

```
#include` "mbed.h"
#include "rtos.h"
 
DigitalOut led(LED1);
 
InterruptIn button(USER_BUTTON);
 
uint32_t button_pressed;
Thread *thread2;
 
void button_press(void)
{
    thread2->signal_set(0x1);
}
 
void led_thread(void const *argument)
{
    while (true) {
        led = !led;
        Thread::wait(1000);
    }
}
 
void button_thread(void const *argument)
{
    while (true) {
        Thread::signal_wait(0x1);
        button_pressed++;
    }
}
 
int main()
{
    Thread thread(led_thread);
    thread2 = new Thread(button_thread);
 
    printf("mbed RTOS example\n");
    
    button_pressed = 0;
    button.fall(&button_press);
    
    while (true) {
        Thread::wait(6000);
        printf("During the last 6 seconds, the Button was pressed %d times\n", button_pressed);
        fflush(stdout);
        button_pressed = 0;
    }
}

bridadan, 2017-01-20:
I just tried the code @jeromecoutant posted in the online compiler with mbed release v134 and mbed-rtos v123 and it worked fine. Same goes for taking it offline and using ARM and uARM toolchians. And just for fun I tried it with mbed OS for the ARM toolchain and it worked as well.

I'll go ahead and close this since it appears to be resolved, please reopen it if you can reproduce the problem.

closed 2017-01-20 by bridadan