ARMmbed/mbed-os #1740: Revision 120 breaks PwmOut on FRDM-K64F

infinnovation, 2016-05-13:
(Copied as requested from https://developer.mbed.org/users/mbed_official/code/mbed/issues/36 )

With revision 120 of mbed, PwmOut on a FRDM-K64F appears to be broken i.e no output is seen.

A demonstration program is at https://developer.mbed.org/users/infinnovation/code/pwm_regression/

On FRDM-K64F, connect LED+330R between PTD0 (J2.6) and GND (J2.3)

With revision 119 of mbed, LED is lit as expected.

With revision 120 of mbed, the LED is not lit.

0xc0170, 2016-05-13:
cc @mmahadevan108

mmahadevan108, 2016-05-16:
Thanks, I will take a look and update.

infinnovation, 2016-05-19:
So, I added some code to examine the FTM registers and found several differences:

FTM3_       |    v119 |     v120 | Comments
----------- | ------- | -------- | --------------
SC          |      8e |       8e |
CNT         |    2958 |      2e2 |
MOD         |    493d |     493d |
C0SC        |      a8 |       a0 | ELSnB:ELSnA=0 => revert pin to GPIO etc.
C0V         |    15f8 |     15f8 |
C1SC        |       0 |        0 |
C1V         |       0 |        0 |
C2SC        |       0 |        0 |
C2V         |       0 |        0 |
C3SC        |       0 |        0 |
C3V         |       0 |        0 |
C4SC        |       0 |        0 |
C4V         |       0 |        0 |
C5SC        |       0 |        0 |
C5V         |       0 |        0 |
C6SC        |       0 |        0 |
C6V         |       0 |        0 |
C7SC        |       0 |        0 |
C7V         |       0 |        0 |
CNTIN       |       0 |        0 |
STATUS      |       1 |        1 |
MODE        |       4 |        5 | FTMEN=1: enable FTM-specific registers
SYNC        |       0 |        8 | SYNCHOM=1: outmask updated by PWM synch
OUTINIT     |       0 |        0 |
OUTMASK     |       0 |        0 |
COMBINE     |       0 | 20202020 | SYNCENn=1
DEADTIME    |       0 |       40 | DTPS=1: (bit ignored if higher bit clear)
EXTTRIG     |       0 |        0 |
POL         |       0 |        0 |
FMS         |       0 |        0 |
FILTER      |       0 |        0 |
FLTCTRL     |       0 |        0 |
QDCTRL      |       0 |        0 |
CONF        |       3 |        3 |
FLTPOL      |       0 |        0 |
SYNCONF     |       0 |     1fb4 | SWSOC,SWINVC,SWWRBUF,SWRSTCNT,SYNCMODE,SWOC,INVC,CNTINC
INVCTRL     |       0 |        0 |
SWOCTRL     |       0 |        0 |
PWMLOAD     |       0 |        0 |

The killer seems to be C0SC where both ELSnA and ELSnB are zero "Pin not used for FTM--revert the channel pin to general purpose I/O or other peripheral control".

I poked 0xa8 into 0x400b900c and, sure enough, my LED came on.

mmahadevan108, 2016-05-19:
At patch with updated FTM, TPM drivers have been submitted to ARM.

infinnovation, 2016-05-19:
OK, thanks.

infinnovation, 2016-06-08:
I can confirm that this issue is fixed in revision 121.

closed 2016-06-08 by infinnovation