ARMmbed/mbed-os #1746: STM32L4 Discovery missing PeripheralPins definitions.

helmut64, 2016-05-16:
@bcostm 
I figured out that serval PINs are missing the in the TARGET_DISCO_L476VG target. My plan was to use SPI2 pins on PD_4, PD_3, PD_1, PE_0, PE_1 which are connected to the onboard LSM303C accelerometer. These pins are not specified in the TARGET_DISCO_L476VG /PeripheralPins.c file and therefore not available using the mbed API's.

Somebody who understands the STM L4 mappings needs to work on this, because I have not sufficient experience with the port mappings. I am happy to test a patch for this.

bcostm, 2016-05-17:
Hi,
We took the decision to write in the PeripheralPins.c file only the pins which are accessibled on a connector (Arduino, Morpho, other...). On the Disco-L476VG board, the pins you mention are not available on the 2 connectors, this is why they are not present.

If you need to drive the LSM303C you can use this example:
[https://developer.mbed.org/teams/ST/code/DISCO_L476VG_Compass/?platform=ST-Discovery-L476VG](url) it uses two libraries (BSP and Compass).

If you need absolutely to control those pins using mbed SPI api (without using above libraries), you need simply to add them in the PeripheralPins.c, that's all.

helmut64, 2016-05-17:
OK, I understand. I am writing an LSM303AGR driver using the mbed API because my final hardware will use this MEMS, for prototyping I swapped out the LSM303C with the LSM303AGR on the L4 Disco board and was hoping to use the mbed SPI API for device access. We prefer to use the "mbed" API instad of the HAL API because it is better documented and we have the flexibility to use all mbed compatible MCU when needed, however I am happy with the L4, great MCU

For the TARGET_DISCO_L476VG /PeripheralPins.c (e.g. PinMap_SPI_SCLK)
I can add:
const PinMap PinMap_SPI_SCLK[] = {
 {PD_1, SPI_2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)},

However I don't know if the "GPIO_AF5_SPI2" selection is correct or if it is "AF6" or "AF7". 

PS: I was hoping when our Hardware is compatible with your L4 Disco board that we can use the mbed build system and deploy it on our own MCU board which is compatible with your "mbed" MCU. From this this point of view to complete definition of all MCU PINs would be appreciated.
Regards Helmut

ohagendorf, 2016-05-17:
Alternate function (AF) 5 is correct. You can find the information in table 16 in L476 datasheet.

helmut64, 2016-05-18:
I was hoping to patch the pin lookup, however it seems that the $Sub$$ does not work with the mbed online compiler. Any other ideas to be able to continue with the online mbed compiler?

Thanks.

#include "pinmap.h"
extern "C" uint32_t $Sub$$pinmap_find_peripheral(PinName pin, const PinMap* map);
extern "C" uint32_t $Super$$pinmap_find_peripheral(PinName pin, const PinMap* map);

const PinMap PinMap_Missing[] = {
    {PD_4, SPI_2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)}, /* MOSI */
    {PD_3, SPI_2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)}, /* MISO */
    {PD_1, SPI_2, STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF5_SPI2)}, /* SCK */
    {NC,    NC,     0}

};

uint32_t $Sub$$pinmap_find_peripheral(PinName pin, const PinMap* map)
{

    uint32_t val;

    val = $Super$$pinmap_find_peripheral(pin, map);
    
    if (val == (uint32_t)NC) {
        if (map->peripheral == SPI_2) {
             const PinMap *m2 = PinMap_Missing;
             while (m2->pin != NC) {
                if (m2->pin == pin)
                    return map->peripheral;
                m2++;
            }   
        }
    }
    return val;
}

bcostm, 2016-05-19:
Just a question: Why you don't send a Pull Request to add these pins in the PeripheralPins.c ? You will have it on mbed IDE on the next mbed release. It's ok with me.

helmut64, 2016-05-19:
I am working on it. First time today, I got the mbed source compiled on my local Windows client, and finally I can debug it using uVision. This makes pull requests much easier because I can test my changes upfront. The pull request will follow.

helmut64, 2016-08-01:
I fixed it and tested it. See #2328

closed 2017-01-16 by sg-