ARMmbed/mbed-os #1754: SPI does not work in revision 120 of mbed.

AnimismRhyme, 2016-05-18:
With revision 120 of mbed, SPI does not work correctly with  FRDM-K64F.

The first byte write worked successfully with correct return(read) value, but following write causes the program hang.

Same project worked fine with previous revision(s) and FRDM-KL25Z.

Tested with following programs.

https://developer.mbed.org/users/Rhyme/code/oscilloscope/

https://developer.mbed.org/users/Rhyme/code/test_spi/

aamcampbell, 2016-05-19:
+1. Ran into this using Neil Thiessen's SDFileSystem.

aamcampbell, 2016-05-19:
FYI, the program hangs in DSPI_MasterWriteDataBlocking(), which is a KSDK 2.0 function (http://kex.nxp.com/apidoc/2.0/group__dspi.html#ga70a0f7d7fe2fbce7993bbcc8c427b2b0). The thing that seems odd to me is that KSDK 2.0 hasn't been released for the K64F, only for the KS22 processor line. I also notice that the file mbed/libraries/mbed/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/TARGET_K64F/spi_api.c is identical to TARGET_K22F/spi_api.c. Is that a mistake? Is it possible that the K64F code is using the wrong implementation of the KSDK libraries, at least for the SPI driver?

loopsva, 2016-05-19:
I have run into other K64F issues with mbed v120. Reference:

https://developer.mbed.org/questions/69021/Questions-about-initialization-failures-/#answer10292?compage=1#c22680

1. Neil Thiessen's SDFileSystem now freezes when doing a disk mount. I am using the SD Flash on the K64F.  Sounds like a related SPI issue.
2. I2C should be seeing  0x3A (the internal FXOS8700CQ) and one external device @ 0x20.  It does not see these intended devices.  Instead, I2C sees different devices (0x10, 0x80 & 0x90)....huh? 
3. Modifying register: UART_PFIFO_REG(UART0) |= 0x09; now creates a compiler error. 
    3a. UART0->PFIFO |= 0x09; compiles ok, but seems to have no effect on the register value.
4. Access to the MII register: uint32_t k64f_mdio_reg = ENET_EIR; now causes a compile error.
    4a. uint32_t k64f_mdio_reg = ENET->EIR; compiles ok, but the register contents never seems to change.  I am looking for the MDIO_MII_READY_BIT in the EIR register to change state.

...kevin

AnimismRhyme, 2016-05-20:
> The thing that seems odd to me is that KSDK 2.0 hasn't been released for the K64F, 
> only for the KS22 processor line. 
Although there are only a few "pre-built" sdks listed, 
we can build/download SDK for K64 from kex.nxp.com. 

I have just built one for myself, and I'm going to check
if the api_spi.c is identical in K22F sdk and K64F sdk.

moto

loopsva, 2016-05-26:
Is there any update on the K64F issues?

AnimismRhyme, 2016-05-30:
I tried find for both SDK_2.0_FRDM-K22F and SDK_2.0_FRDM-K64F and there was no spi_api.c.
So I can not judge if the difference/similarity of spi_api.c for these platforms was the cause of the problem.
For my side, I just down-graded the mbed lib to 119 for FRDM-K22F and have managed to survive so far. ;-)

0xc0170, 2016-05-30:
cc @mmahadevan108 Please have a look

The layout is not the best as the moment. Some files might be the same. In this case K22 and K64F might have same spi peripheral, but KL27Z does not, and all of these are under ksdk. It might be better to have the implementation based on the peripherals. In this case, it's dspi vs spi. Thus dspi driver would be used by nxp chips which contain dspi peripheral. this is however not a problem of this issue, we should discuss this in another one. I'll talk to Mahesh about it.

Could you please run spi test here in the codebase ? I'll check hte test resutl, I did not notice SPI test fails for k64f. 

cc @bridadan - hw tests for k64f :-)

AnimismRhyme, 2016-05-30:
As I wrote, the first byte write and read works fine, therefore it might have been difficult to capture the problem unless you ran multiple transactions test. Meantime, since the first byte write/read worked, it took us rather long time to figure out the cause of hang was SPI, instead of our application code.

0xc0170, 2016-05-31:
This PR should fix it - #1817

AnimismRhyme, 2016-06-01:
I have just tested my test_spi program on FRDM-K64F with mbed lib 121.
And alas, it did not work.
With same setup, if I down-grade the lib to 119, it starts working again.
So I'm afraid that currently mbed lib 120 and mbed lib 121 are NG (for FRDM-K64F spi).

0xc0170, 2016-06-01:
Yes, you can use mbed-dev and update SPI based on the fix I referenced above. The fix will be part of the next mbed release.

0xc0170, 2016-06-16:
We shall close it, if you think it's still present on the master, please reopen

closed 2016-06-16 by 0xc0170