ARMmbed/mbed-os #1756: Exporter for e2 studio

phyokyaw, 2016-05-19:
We contributed e2 studio toolchain exporter in #1659, it's merged many thanks for that. 

Not sure this is the right place to raise an issue.

the exporter is working file, just wondering if it is possible to update the exporter drop download list with

1. e2 studio icon (attached)
2. Change name from E2studio to e2studio

Many thanks.

Phyo

[image]

phyokyaw, 2016-06-13:
Any update on this please.

0xc0170, 2016-06-13:
cc @screamerbg

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-