ARMmbed/mbed-os #1770: Bug in I2C implementation in the SAMG55J18

kgoveas, 2016-05-24:
Hi @Parthasarathy,

I discovered an implementation bug in the i2c_api.c. The address in line 270 and line 358 should not be modified as it ends up corrupting the address. 
Current code
`packet.chip=(address>>1) & 0x7F;`
Expected code
`packet.chip=address;`

Parthasarathy, 2016-05-26:
Hi @kgoveas 

Please refere the following link https://developer.mbed.org/handbook/I2C
The Note says "The mbed API uses 8 bit addresses, so make sure to take that 7 bit address and left shift it by 1 before passing it"

The code handles the use case converts the address back to 7 bit address.

Thanks,
Partha

kgoveas, 2016-05-27:
@Parthasarathy ,

Thank you for info.

Kind Regards,
Kevin

closed 2016-05-27 by kgoveas