ARMmbed/mbed-os #1779: Exporting the mbed SDK causes the HTTP request to hang

Bitbored, 2016-05-25:
When trying to [export the mbed SDK](https://developer.mbed.org/users/mbed_official/code/mbed/export) the request hangs and eventually returns a 504 gateway error, never sending the expected zip file. This has also been reported on the [issue tracker of the mbed website as issue 35](https://developer.mbed.org/users/mbed_official/code/mbed/issues/35), but no response was ever given.

I mostly tested this with the LPC1768 and EFM32 USB-enabled Happy gecko platforms and the GCCARM and Zip archive toolchains, but no combination seems to work.

Is there an other way to get a specific build, as described in the mbed.bld file of a project, of the mbed SDK for a specific platform without using the online compiler? Checking out the complete mercurial repo or downloading the zip file of the prebuilt SDK is pretty inefficient since it contains binaries for all the available targets.

0xc0170, 2016-05-25:
That issue tracker should be disabled, or just redirected here. Anyway, this relates to website. There's https://developer.mbed.org/forum/bugs-suggestions/. 

I closed the issue under mbed lib you referenced above.

cc @screamerbg @stephenpaulger Could you please help out?

Bitbored, 2016-05-25:
There seems to be a [report on this issue on the forum as well](https://developer.mbed.org/forum/bugs-suggestions/topic/26128/). There might be more, but I have not found out how to search on that forum yet...

According to @sg- this [should have been fixed about a moth ago](https://developer.mbed.org/questions/68709/Export-to-ZIP-not-working/), but it is still very much broken.

0xc0170, 2016-06-01:
I can confirm, I am getting for zip 504 gateway error. However this is not website tracker. I notified the web team, and for the future, please report it here https://developer.mbed.org/forum/bugs-suggestions/

closed 2016-06-01 by 0xc0170

Bitbored, 2016-06-01:
It has been reported here, even though IMO the description is not that clear: https://developer.mbed.org/forum/bugs-suggestions/topic/26128/