ARMmbed/mbed-os #1786: Along with the version up of "lwip-eth" library, WebCamera program can not be compiled.

TomoYamanaka, 2016-05-26:
Hi,

Along with the version up of "lwip-eth" library, WebCamera program can not be compiled.
&nbsp; &nbsp; https://developer.mbed.org/teams/Renesas/code/GR-PEACH_WebCamera/

The compile error is that there is no file "lwipopts_conf.h".
WebCamera program uses the "EthernetInterface" library. At "lwipopts.h" in "lwip" library called from "EthernetInterface" library, "lwipopts_conf.h" file are included, where it was an error.

At "lwipopts_conf.h" in "lwip-eth" library called from "EthernetInterface" library, this file seems to have been deleted by version-up of "lwip-eth" library two-days ago.
&nbsp; &nbsp; https://developer.mbed.org/users/mbed_official/code/lwip-eth/rev/3534f0b45d38

We consider that also the changes to "lwipopts_conf.h" file is necessary. how is it?

Regards,
Yamanaka

0xc0170, 2016-05-26:
cc @screamerbg

0xc0170, 2016-05-26:
The pull request did not touch the lwip files, therefore I suspect the sync script is one to blame, we will investigate

0xc0170, 2016-05-26:
I spot at least one error in the sync script, I'll send PR to fix it: https://github.com/mbedmicro/mbed/commit/9db0f2e6e09a1dbe1328b445233f595b9928e7ed#diff-cc93abe087d6d55305ba6709ea7b5043L59

0xc0170, 2016-05-26:
This was automatically fixed via gitbhub. The sync script was fixed, the library should be sync and the files should be back.

closed 2016-05-26 by 0xc0170

0xc0170, 2016-05-26:
Fixed - https://developer.mbed.org/users/mbed_official/code/lwip-eth/rev/9de8bd8ca1c8 ! Please report back if not

TomoYamanaka, 2016-05-26:
Thank you for fix.
WebCamera program could be compiled.

Regards,
Yamanaka

PR #1787 (2016-05-26): Synch - fix lwip-eth path
Fixes #1786 issue. lwip-eth is located in the ethernet sources ,not lwip (will be fixed later)

@screamerbg
commit 627fe55c (2016-05-26): Synch - fix lwip-eth path

Fixes #1786 issue