ARMmbed/mbed-os #1794: Very little cross-platform documentation

AngusP, 2016-05-26:
All of the included documentation and the vast majority of online documentation focuses entirely on Windows use - for example, I'm pretty lost as to what, exactly, I as a Linux user need to put in `private_settings.py` to get the SDK functioning, with the [GCC ARM embedded](https://launchpad.net/gcc-arm-embedded) toolchain. I'd also assume that there are developers that would appreciate documentation for Macs.

I'd be more than willing to contribute documentation, but need a little help getting past this hurdle given I've been fiddling with this for a while

theotherjimmy, 2016-05-26:
TL;DR: Nothing! (I don't even have that file) You simply need to use the `GCC_ARM` target (`-t` option) when building after installing the tool-chain as described in the link you provided.

More details than are needed: The settings file (`workspace_tools/settings.py`) creates a few reasonable default settings and then tries to import the settings from a private settings file (it assumes that the file will be named `workspace_tools/private_settings.py`). The default search path for the `GCC_ARM` target is Nothing. This forces the tools to search you `PATH` environment variable for the executables it will use (`arm-none-eabi-{gcc,g++,ld,objdump,size}`). The Ubuntu packages provided by the `GCC ARM` embedded add these executable to `/usr/bin`, which should already be part of your `PATH`.

AngusP, 2016-05-27:
Having had more of a fiddle, after pulling and installing [gcc-arm-embedded](https://launchpad.net/gcc-arm-embedded) you do need to fiddle with the settings files. So far the following has worked (given I'm on fedora 22, but this should work for the debian derivatives too)

This is mostly because installing the entire GCC ARM toolchain system-wide has a bunch of distro-specific caveats, so the downloaded binary install ends up being easier for most.

```python
# workspace_tools/private_settings.py
from os.path import join

ARM_PATH = "<wherever you put it>/gcc-arm-none-eabi-5_3-2016q1"
ARM_BIN = join(ARM_PATH, "bin")
ARM_INC = join(ARM_PATH, "include")
ARM_LIB = join(ARM_PATH, "lib")

GCC_ARM_PATH = join(ARM_PATH, "bin")
```

Then `(venv)> python ./workspace_tools/build.py -t GCC_ARM` worked fine.

0xc0170, 2016-05-27:
@AngusP Send a patch for https://github.com/mbedmicro/mbed/blob/master/docs/BUILDING.md#configure-workspace-tools-to-work-with-your-compilers, if you consider there are some pieces missing. 

ARM_PATH is for ARMCC, not gcc arm as stated in the doc above

0xc0170, 2016-06-01:
Is there anything else we can do to resolve this issue?

bridadan, 2017-01-18:
This is currently documented on the mbed CLI page: https://github.com/ARMmbed/mbed-cli#compiling-code

All updates to the docs should happen there. I'm going to close this for now, if you'd like us to clarify this further please reopen!

closed 2017-01-18 by bridadan