ARMmbed/mbed-os #1798: In few cases STM32 UART tx might not complete

LMESTM, 2016-05-27:
In rare racing cases, the uart transmission may not complete.
Such case are for instance direct reconfiguraiton of UART after printf, or immediate deepsleep entry after printf.  There is a fix ongoing, together with a proposed new test case to avoid later regression.

sg-, 2017-01-16:
@LMESTM Is this resolved?

LMESTM, 2017-01-16:
@sg- how may I check status of ARM Internal Ref: IOTMORF-180 ? 

I don't think this has fully been addressed. My proposed fix in STM targets was rejected as this would reduce flexibility, but I have not seen a proposal at MBED design level - for now I think this is left to application to leave enough time for buffer to empty.

sg-, 2017-01-16:
> @sg- how may I check status of ARM Internal Ref: IOTMORF-180 ?

Its closed. We are tracking this here for now.

LMESTM, 2017-01-16:
@sg ok. 
I mentioned, I don't think that this has been fully addressed, but if there are no pending actions, then better close it as I won't do anything more on my side either for now

closed 2017-04-27 by LMESTM