ARMmbed/mbed-os #1809: Internal fault: [0xc43ed8:5050106] in osMailGet

cyliangtw, 2016-05-30:
To build RTX lib and ethernet lib by
  `python ./workspace_tools/build.py -c -m LPC1768 -t ARM -r -e ` 
The, suffered the internal fault.
Log show as:
```
...
Compile: RTX_Conf_CM.c
Compile: HAL_CM3.c
Compile: SVC_Table.S
Compile: rt_CMSIS.c
Internal fault: [0xc43ed8:5050106] in osMailGet
Please contact your supplier.
```

0xc0170, 2016-05-30:
@cyliangtw Please report the version of ARMCC you are using please. I recall some internal fault previously, which was fixed. We should check the git log to see if this is still the case.

cyliangtw, 2016-05-30:
Hi Martin,
ARMCC version: MDK-ARM V5.14.0.0 .

cyliangtw, 2016-05-31:
Hi Martin, 
If I rollback ARMCC version as MDK-ARM V4.74, I could build RTOS & ethernet lib successfully.

embeddedartists, 2016-05-31:
I had the same problem (was using MDK-ARM 5.14.0.0) and read somewhere that it was solved in version 5.18.0.0. I have upgraded to the latest version (5.20.0.0) and no longer see the problem.

toyowata, 2016-05-31:
>Internal fault: [0xc43ed8:5050106]

This is probably a compiler bug which is fixed in ARM Compiler 5.06.  The defect number was SDCOMP-25465.

http://infocenter.arm.com/help/topic/com.arm.doc.arn0005v/index.html#history

>[SDCOMP-25465]  When compiling for Thumb state with -Otime at -O2 or above, code containing a comparison with a number that cannot be used as an immediate with the ADDS instruction could cause the compiler to report an Internal fault. This has been fixed.

So, you will not see this internal fault if you use MDK-ARM v5.17 or later.

|MDK-ARM|ARM compiler|
|:-----------|------------:|:------------:|
|v5.14|v5.05 update 1|
|v5.15|v5.05 update 2|
|v5.16|v5.05 update 2|
|v5.17|v5.06 update 1|
|v5.18|v5.06 update 1|
|v5.20|v5.06 update 2|

cyliangtw, 2016-05-31:
Got it & thanks.

closed 2016-05-31 by cyliangtw