ARMmbed/mbed-os #1819: Overwrite _sbrk() to support external SRAM

ccli8, 2016-05-31:
Hi,
I am porting mbed classic to Nuvoton's Nu-mbed-NUC472 board. There is external SRAM on this board. To locate heap on the external SRAM, I need to overwrite _sbrk(). But I meet 'multiply defined' error on GCC_ARM target. The _sbrk() implementation in retarget.cpp cannot meet my requirement (heap and stack are not contiguous in my case). Could _sbrk() in retarget.cpp declare to be weak to overwrite for like my case?
https://github.com/mbedmicro/mbed/blob/master/hal/common/retarget.cpp

0xc0170, 2016-06-01:
cc @niklas-arm

niklas-arm, 2016-06-01:
Short of modifying the code to attribute that function to be `weak`, you may be able to wrap this function using the GNU linker by `-Wl,--wrap,_sbrk`.

Please note, that `dlmalloc` assumes a linear heap address space, meaning with `_sbrk` you can only have _one_ heap section and it must be in continous memory.
On Cortex-A this isn't an issue, because the MMU maps non-continous memory (ie. pages) to appear linear, but on Cortex-M you do not have this functionality, and the simple `_sbrk` approach does not work anymore.

For you this mean, you'd have to relocate your _entire_ heap to the external memory, you cannot use the internal SRAM anymore, because that would require you to have two memory pools, and `_sbrk` and `dlmalloc` cannot deal with that.

Also note that external RAM may consist of several banks and it may not be possible to do unaligned access across bank address boundaries. Though I'd expect the external memory peripheral to take care of splitting unaligned accesses into two aligned accesses, but that depends on the vendor implementation.
In the worst case, your external heap `_sbrk` is limited to use only _one_ bank of the external memory, due to this issue.

niklas-arm, 2016-06-01:
For clarity: 

> you cannot use the internal SRAM anymore

for heap usage. Of course, it's still usable for stack and static data and whatever else you need.

ccli8, 2016-06-02:
@niklas-arm In my scenario, there is just one bank of external SRAM in the Nu-mubed-NUC472 board. I can relocate the whole heap to the external SRAM now per your suggestion. Thanks for your information.

ccli8, 2016-07-11:
@niklas-arm In original workspace_tools/targets.py, I can wrap `_sbrk` (`-Wl,--wrap,_sbrk`) by `hook.hook_cmdline_linker`. But after target definition is moved from workspace_tools/targets.py to hal/targets.json, I find no way to adding such hook. Do you have a plan to support such hook function?
```
class NUMBED_NUC472(Target):
    def __init__(self):
        Target.__init__(self)
        self.core = "Cortex-M4F"
        self.is_disk_virtual = True
        self.extra_labels = ['Nuvoton', 'NUC472', 'NUMBED_NUC472']
        self.macros = ['TARGET_NUVOTON', 'TARGET_NUC472', 'TARGET_NUMBED', 'TARGET_NUC472HI8AE']
        self.supported_toolchains = ["GCC_ARM", "ARM", "uARM", "IAR"]
        self.default_toolchain = "ARM"
        # NOTE: Pin mappings of Arduino form factor are inconsistent with test samples spi_master/spi_slave, so cancel it.
        #self.supported_form_factors = ["ARDUINO"]
        self.detect_code = ["1301"]
        self.progen = {
            "target": "numbed-nuc472",
        }
    
    def init_hooks(self, hook, toolchain_name):
        if toolchain_name in ['GCC_ARM']:
            hook.hook_cmdline_linker(self.cmdline_linker_hook)
    
    def cmdline_linker_hook(self, toolchain_name, cmdline):
        if toolchain_name in ['GCC_ARM']:
            # Wrap calls to _sbrk() to relocate heap on external SRAM.
            cmdline.append(r'-Wl,--wrap,_sbrk')
        return cmdline
```

niklas-arm, 2016-07-11:
cc @0xc0170 @bogdanm @sg- for workspace tools

ccli8, 2016-12-12:
On mbed OS 5. the wrap mechanism cannot work. I overwrite `_sbrk` with conditional compile in `retarget.cpp`. This modification has merged and so I close the issue.

closed 2016-12-12 by ccli8