ARMmbed/mbed-os #1823: Support STM32F769I Discovery for mbed

sandeepach, 2016-05-31:
Support STM32F769I Discovery for mbed. This board support double precision floating point which i want to use with mbed.

0xc0170, 2016-06-01:
@sandeepach What about DISCO_F746NG ?

cc @bcostm @adustm

ohagendorf, 2016-06-01:
FPU of DISCO_F746 is single precision
FPU of DISCO_F769 is double precision

adustm, 2016-06-02:
Hello, no target date for this platform at the moment.
Cheers

sandeepach, 2016-06-03:
As a part of ohagendorf's changes for support Nucleo F767, isnt it easy from there to support Disco 769? 
I think even Disco 769 is based on Cube 1.4.0. 
I will be mainly interested in utilizing the 4 MEMS Microphones there on that board.

ohagendorf, 2016-06-03:
I can add that target, no problem.

sandeepach, 2016-08-16:
Any chance of this happening anytime soon.

ohagendorf, 2016-08-16:
It is almost ready to submitt. I'll do it during the next days.

sg-, 2017-01-16:
Looks like Disco STM32F769I is in the codebase.

sandeepach, 2017-01-16:
It is in the codebase but not in the list of supported platforms or even in mbed online compiler. 
Request to reopen the issue.

sg-, 2017-01-16:
@adustm Suggested no target date. Will reopen for the request.

adustm, 2017-01-16:
Hello, ST team has just submitted some modifications compared to the currently available platform. 
https://github.com/ARMmbed/mbed-os/pull/3571
Once this PR will be made available in an mbed official release, we will enable it in the online compiler.

Cheers

bridadan, 2017-01-20:
Closing this as #3571 was merged, thanks @adustm and @jeromecoutant!

closed 2017-01-20 by bridadan