ARMmbed/mbed-os #1834: mbed lib 121 AnalogIn returning int16_t instead of uint16_t

AnimismRhyme, 2016-06-02:
This morning I was preparing some sample project for FRDM-K22F 
which utilizes AnalogIn and PwmOut. 

As I have already annoyed with the SPI problem in mbed lib 120 
I was not surprised when I saw the values from AnalogIn was screwed. 

It seems that the read_us16() actually returning signed 16bit value. 
and read() goes with it, and resulting very interesting values. 

So I just re-winded the mbed lib to 119 and it resumed working fine.

Following is the read of AnalogIn with a pot attached to FRDM-K22F and 
started from 0 to max slowly. 
So the values are expected to be starting 0 (0x0000) to 1.0 (0xFFFF). 

mbed lib 119
0.000000 0x0000
0.176410 0x2D29
0.316960 0x512D
0.394812 0x6508
0.628733 0xA0EE
0.745464 0xBF08
0.870832 0xDEEB
0.999664 0xFFE9
0.999649 0xFFE9
0.999756 0xFFEE
0.999771 0xFFEC
0.999695 0xFFEA
0.999710 0xFFEC

mbed lib 121
0.000000 0x0000
0.998688 0xFFA8
0.999435 0xFFDA
0.042756 0x0AFD
0.075425 0x134F
0.118608 0x1E5F
0.135149 0x229B
0.180850 0x2E44
0.222110 0x38DF
0.261646 0x4300
0.301335 0x4D17
0.342687 0x57B0
0.388235 0x6362
0.492363 0x7E0B

sigh
moto

0xc0170, 2016-06-02:
Looking at the sources, the analogin_read_u16 for KSDK2 MCU returns the result from this function ``uint32_t ADC16_GetChannelConversionValue(ADC_Type *base, uint32_t channelGroup)``.

Referenced line: https://github.com/mbedmicro/mbed/blob/master/hal/targets/hal/TARGET_Freescale/TARGET_KSDK2_MCUS/api/analogin_api.c#L78

0xc0170, 2016-06-02:
cc @mmahadevan108

AnimismRhyme, 2016-06-02:
> ADC_Type structgure contains R register (two of them), which are defined as uint32_t , 
> thus return of the function ADC16 referenced above is uint32_t.

In 35.3.4 ADC Rata Result Register (ADCx_Rn) of K64P144M120SF5RM,
the lower 16bit of R register holds either signed or unsigned 16bit value,
along with all 0 in the upper 16bit.

So there can be plenty of chance that the 16bit was signed,
but reading your analogin_api.c at least the resolution was assigned as
    adc16_config.resolution = kADC16_ResolutionSE16Bit;
so I hope that the result 16bit should be unsigned.

Another chance to be miss-align of data can sneak in might be
    return ADC16_GetChannelConversionValue(adc_addrs[instance], 0);
here, you are returning 32bit value to 16bit value, but as it is taken as uint16_t
I don't understand where this signed stuff sneaked in.

Anyway my test result is suggesting that the read values were signed somehow.

AnimismRhyme, 2016-06-03:
I modified my program to display the ADC0 Result Register directly, too.
And the value of read_u16() and the register value were same.
So somehow, the ADC module is setup for signed 16bit instead of
unsigned 16bit somehow/somewhere.

0.000000 0x0000 : 0x00000000
0.998672 0xFFA9 : 0x00000000
0.998703 0xFFA9 : 0x0000FFA9
0.029465 0x0788 : 0x0000FFA9
0.063310 0x1032 : 0x00000788
0.087938 0x1681 : 0x00001032
0.111360 0x1C7E : 0x00001681
0.155352 0x27C2 : 0x00001C7E
0.196246 0x323C : 0x000027C2
0.223896 0x394C : 0x0000323C
0.290135 0x4A45 : 0x0000394C
0.324910 0x5334 : 0x00004A45
0.401572 0x66C9 : 0x00005334
0.492363 0x7E0B : 0x000066C9
0.492363 0x7E0B : 0x00007E0B
0.492363 0x7E0B : 0x00007E0B

AnimismRhyme, 2016-06-03:
I ran additional test to check the value of ADC0_SC1 and ADC0_CFG1
ADC0_SC1 = 0x0000_0028
ADC0_CFG1 = 0x0000_006C
This means that ADC was configured as
ADC0_SC1[DIFF] = 1
 Differential conversions and input channels are selected
ADC0_SC1[ADCH] = 01000 
 When DIFF=0, AD8 is selected; When DIFF=1 it is reserved
ADC0_CFG1[MODE] = 11 
 When DIFF=0: it is single-ended 16-bit conversion
 When DIFF=1: it is differential 16-bit conversion with 2's complement output

So, my conclusion is that somewhere/somehow ADC0_SC1[DIFF] was set
which is actually illegal setup for AD8. 
My best guess is that it is taking place in the constructor of AnalogIn(port_name).

AnimismRhyme, 2016-06-03:
Well, I was wrong...
In my first part of the program I assigned 0x0000_0008 to ADC0_SC1 and immediate print out was displaying the right value. But next write shows the value was changed to 0x0000_0088 and
after read_u16() it was changed to 0x0000_0028 and it stays as 0x0000_0028 rest of my program.
I wonder why but some other part of the program is modifying the ADC register...

mmahadevan108, 2016-06-03:
Can you please send your test program for me to replicate the issue and test any fix.

AnimismRhyme, 2016-06-04:
/* current test program of mine */
#include "mbed.h"

AnalogIn a0(PTB0) ;
PwmOut led(PTD5) ;

uint32_t *adc_result = (uint32_t*)0x4003B010 ; /* ADC0_RA */
uint32_t *adc_screg = (uint32_t*)0x4003B000 ; /* ADC0_SC1 */
uint32_t *adc_config = (uint32_t*)0x4003B008 ; /* ADC0_CFG1 */
int main()
{
    float fvalue ;
    uint16_t uvalue ;
    uint32_t result_reg1 ;
    uint32_t result_reg2 ;
    *adc_screg = 0x00000008 ;
    printf("\nadc_screg = 0x%08X\n", *adc_screg) ;
    while (true) {
        result_reg1 = *adc_screg ;
        result_reg2 = *adc_config ;
        fvalue = a0.read() ;
        uvalue = a0.read_u16() ;
        led = fvalue ;
        printf("%f 0x%04X : 0x%04X 0x%04X\n",
            fvalue, 
            uvalue, 
            result_reg1 & 0xFFFF, 
            result_reg2 & 0xFFFF) ;
        wait(1) ;
    }
}
/* end current test program of mine */

AnimismRhyme, 2016-06-04:
lib 121 test result
adc_screg = 0x00000008
0.998672 0xFFA8 : 0x0088 0x006C
0.998688 0xFFA8 : 0x0028 0x006C
0.022995 0x05E4 : 0x0028 0x006C
0.033249 0x088C : 0x0028 0x006C
0.056977 0x0E94 : 0x0028 0x006C
0.074449 0x1310 : 0x0028 0x006C
0.111177 0x1C7F : 0x0028 0x006C
0.153689 0x2757 : 0x0028 0x006C
0.181613 0x2E80 : 0x0028 0x006C
0.218204 0x37E8 : 0x0028 0x006C
0.290028 0x4953 : 0x0028 0x006C
0.443519 0x7198 : 0x0028 0x006C
0.492363 0x7E0B : 0x0028 0x006C
0.492363 0x7E0B : 0x0028 0x006C

AnimismRhyme, 2016-06-04:
lib 119 test result
adc_screg = 0x00000008
0.000000 0x0000 : 0x0088 0x006C
0.000122 0x0009 : 0x0008 0x006C
0.032883 0x0862 : 0x0008 0x006C
0.051698 0x0D38 : 0x0008 0x006C
0.070008 0x11E9 : 0x0008 0x006C
0.127214 0x2092 : 0x0008 0x006C
0.208225 0x3544 : 0x0008 0x006C
0.326177 0x5378 : 0x0008 0x006C
0.388205 0x635C : 0x0008 0x006C
0.483146 0x7B9F : 0x0008 0x006C
0.658488 0xA888 : 0x0008 0x006C
0.769314 0xC559 : 0x0008 0x006C
0.999634 0xFFE5 : 0x0008 0x006C
0.999725 0xFFEF : 0x0008 0x006C
0.999817 0xFFF2 : 0x0008 0x006C
0.999680 0xFFED : 0x0008 0x006C

mmahadevan108, 2016-06-07:
It looks like the DIFF bit is getting incorrectly set. I will make a change to explicitly set this option to false in the config parameters. That should fix this issue. Thanks.

mmahadevan108, 2016-09-19:
Can we close this issue?

AnimismRhyme, 2016-09-20:
Dear mmahadevan108-san,

I have just tested my program with mbed-lib v126
and it is working fine!
So I think that we can call this issue complete and 
should be able to be closed.

Best Regards,
20-Sep-2016
Motoo Tanaka

closed 2016-09-20 by AnimismRhyme