ARMmbed/mbed-os #1841: critical section - rtos problematic

0xc0170, 2016-06-02:
Our current implementation of critical section uses ``__disable_irq``, which looks fine for bare-metal.

The problem becomes when rtos is included. To illustrate:

```
// rtos present in this code
critical_section_enter();
printf("Hello\n");  // just for illustration, I could invoke here some RTOS code instead , the same result
critical_section_exit();
```

This fails for ARMCC as stdlib requires locking which is part of RTX (mutex for stdlib mutex_require/release calls). Therefore printf tries to acquire lock, but because interrupts are disabled (let's assume cortex-m0, leave BASEPRI out of the picture, we care about generic use case - any cortex). printf ends up in the hard fault. Because when mutex is called, it does ``SVC 0x00``, this causes a hard fault.
h
We should consider RTOs version of critical section, it does not make sense there to disable interrupts anyway

@adbridge Thanks for the help

cc @c1728p9 @sg-

Patater, 2016-06-14:
Additionally, the assertions inside `core_util_criticial_section_enter` and `core_util_critical_section_exit` fail when the RTOS is running in `#define OS_RUNPRIV 0` mode, because from unprivileged mode, writes to `PRIMASK` are ignored and reads from `PRIMASK` are always zero (on Cortex-M3-like systems).

When these assertions fail, we observe infinite recursion, as `mbed_assert_internal` also calls `core_util_critical_section_enter`, which then fails an assertion and calls `mbed_assert_internal`, which then calls `core_util_critical_section_enter`, and so on.

0xc0170, 2016-06-14:
@Patater Any suggestions to fix that recursion, plus assertions with RTOS in unprivileged mode?

closed 2017-10-27 by ChiefBureaucraticOfficer