ARMmbed/mbed-os #1842: Timer jumps after 35 minutes on K64F

marcuschangarm, 2016-06-02:
When running the following program on a K64F, the timer value jumps drastically after 35 minutes.

```C++
#include "mbed.h"
#include "rtos.h"

Timer tick;

int main()
{
    tick.start();

    for(;;)
    {
        printf("%lu\r\n", tick.read_ms());
        wait(1);
    }
}
```

Output after 35 minutes:

```
2144601
2145608
2146615
4292819952
4292820963
4292821973
```

0xc0170, 2016-06-02:
I would like to test this on earlier revision (mbed lib <120) when we used KSDK 1.0

cc @mmahadevan108

mmahadevan108, 2016-06-07:
Hi Martin, Does this exist on the earlier version as well?

0xc0170, 2016-06-07:
@marcuschangarm What version of mbed are u running? SHA? Can u test the older version with KSDK?

marcuschangarm, 2016-06-08:
I'm on #974ecf731c8a5de78acc9e40573f2a7c7641c2cd.

What version do you want me to test?

0xc0170, 2016-06-08:
@marcuschangarm https://github.com/mbedmicro/mbed/commit/868ca35dfb02565beafef75f12e5f0f1740e860c (use mbed lib 119) please and let us know

marcuschangarm, 2016-06-08:
Same result on mbed 119 and mbed-rtos 108:

```
2144600
2145608
2146615
4292819952
4292820963
4292821973
```

svastm, 2016-06-16:
It's an integer overflow error. `2 146 615` ms mean `2 146 615 000` us which is just a little bit under the max integer (32bit) value. We could had one of this behavior to handle this type of overflow:
- reset the timer
- return an overflow error code

If you need a clock you can use the rtc (if your device has one).

marcuschangarm, 2016-06-16:
@svastm thank you for the explanation. 

I'm a bit surprised the Timer API uses signed values for measuring time. 

In this particular case I would have expected the timer to reset on an overflow.

pan-, 2016-10-18:
@0xc0170 Any update on this ? 
It is an API defect affecting all targets and every release of mbed-os. 
What should be done ?

marcuschangarm, 2016-10-19:
As a user, I would expect that:
- reading a timer would return an unsigned value.
- I could register a callback function to receive overflow callbacks.
- a function called read_ms would have increased range as opposed to read_us, i.e., read_us should overflow 1000 times before read_ms would.

pan-, 2016-10-19:
> I could register a callback function to receive overflow callbacks.

What is the purpose ? If an overflow occur, the user won't see it until its overflow callback is called. During this time, the timer can be read and the overflow won't be handled. 

Checking overflow before (or after) the timer read induce the same kind of issue. Again, there is a small window of time between the timer read operation and the overflow check which can cause an improper handling of an overflow.

marcuschangarm, 2016-10-19:
@pan- the purpose would be to track overflows and extend the current timer to higher precision. E.g. I could increment an overflow counter in the callback and extend the timer beyond what is currently supported.

sg-, 2017-01-16:
This is documented. https://docs.mbed.com/docs/mbed-os-api-reference/en/latest/APIs/tasks/Timer/ We should open a new issue to create a unified 64bit timestamp

marcuschangarm, 2017-01-16:
@sg- What I'm missing from the docs is that read_ms and read_us are based on the same variable and will overflow at the same time. As a user, I would expect the read_ms to overflow on MAX_INT32 and less frequent than read_us.

sg-, 2017-01-16:
OK - the warning jumped out at me but we can reopen and update the doxygen API docs to be more clear

marcuschangarm, 2017-01-16:
Thank you! (what timezone are you in? :smile:)

closed 2017-10-27 by ChiefBureaucraticOfficer