ARMmbed/mbed-os #1857: Issues after device.h -> targets.json conversion

bogdanm, 2016-06-06:
There are still some `device.h` files with actual data left in the tree, see for example https://github.com/mbedmicro/mbed/blob/master/hal/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/TARGET_NRF51822_MKIT/device.h. These should have gone away.
Cc @theotherjimmy

theotherjimmy, 2016-06-06:
you're right. these should have gone away. I missed the warning: 
[WARNING] device NRF51822_MKIT did not have an associated entry in targets.json

0xc0170, 2016-06-06:
Please @theotherjimmy send a fix or let us know otherwise

theotherjimmy, 2016-06-06:
Working on it. will link this issue with the pr when I get that up.

theotherjimmy, 2016-06-06:
so, the reason for the warning is that there is no entry with the string "NRF51822_MKIT" as part of the name or the extra labels of any entry within target.json. should I instead be matching on the MCU_NRF51822 instead?

theotherjimmy, 2016-06-06:
https://github.com/mbedmicro/mbed/pull/1859

theotherjimmy, 2016-06-06:
@bogdanm  the PR was merged. is this all cleaned up now?

theotherjimmy, 2016-06-09:
@bogdanm @0xc0170 PING! Is this resolved?

0xc0170, 2016-06-09:
It should be

closed 2016-06-09 by 0xc0170