ARMmbed/mbed-os #1861: GitHub doesn't appear to be syncing to developer.mbed.org

neilt6, 2016-06-06:
@0xc0170 Changes to the GitHub codebase don't appear to syncing to the libraries on https://developer.mbed.org/ as of a few weeks ago when everything was reorganized.

0xc0170, 2016-06-06:
Synch was disabled right before that reorganization, we need to synch a releases not directly commits, and libraries should be in synch with the rest of the code  (tools, hal, cmsis).

Thanks for reporting, lets keep this open until resolved with a proper solution.

0xc0170, 2016-06-07:
cc @sg-

neilt6, 2016-06-21:
@0xc0170 Any update on when sync will be re-enabled? I'm interested in using the new `Callback` API that was added to the RTOS library.

neilt6, 2016-07-25:
Just noticed that the [mbed](https://developer.mbed.org/users/mbed_official/code/mbed/) and [mbed-rtos](https://developer.mbed.org/users/mbed_official/code/mbed-rtos/) libraries were updated this morning. That's awesome! Is the [mbed-dev](https://developer.mbed.org/users/mbed_official/code/mbed-dev/) library going to be updated too? Thanks guys!

0xc0170, 2016-07-25:
mbed lib rev 122 was published along  with the rtos update.  Please see the tag what commit they refer to.

neilt6, 2016-07-26:
@0xc0170 - To clarify, the [pre-built mbed library](https://developer.mbed.org/users/mbed_official/code/mbed/) was updated yesterday, but the [mbed source library](https://developer.mbed.org/users/mbed_official/code/mbed-dev/) hasn't been updated yet. I use the source library in my projects so that I can tweak things and disable some of the debugging code by defining `NDEBUG`.

sg-, 2016-07-26:
Hi @neilt6 working to update that and improve the experiences. Hang in there should have something for you soon!

neilt6, 2016-07-26:
Thanks @sg- !

neilt6, 2016-10-11:
Is it possible to get a sync on [USBDevice](https://developer.mbed.org/users/mbed_official/code/USBDevice/) and [USBHost](https://developer.mbed.org/users/mbed_official/code/USBHost/)? They've been updated here recently for better compatibility with the new mbed classic libraries.

sg-, 2017-01-16:
We'll be manually synchronizing USB and cleaning up developer site pull requests until integrated with mbed OS 5.

closed 2017-01-16 by sg-