ARMmbed/mbed-os #1862: Building mbed library for Nucleo_L152RE

hesolium, 2016-06-06:
I have a problem with the offline creation 'mbed' library for board Nucleo_L152RE.
Compilings runs correctly (no errors or warnings). Library (libmbed.a) is created but the file 'devices.h' in folder '/.build/mbed/TARGET_NUCLEO_L152RE/TARGET_STM/TARGET_STM32L1/TARGET_NUCLEO_L152RE' is basically empty. There are not #defines (eg. DEVICE_SERIAL, ...). An attempt to compile an application fails with error: 'Serial is not a type.'
I create a library as usual with the command: 'python workspace_tools/build.py -m NUCLEO_L152RE -t GCC_ARM -o "debug-info"'.

0xc0170, 2016-06-06:
cc @theotherjimmy

theotherjimmy, 2016-06-06:
I can't reproduce the error. running``` python workspace_tools/build.py -m NUCLEO_L152RE -t GCC_ARM -o "debug-info" ``` works.

theotherjimmy, 2016-06-06:
right. you want to make sure that your build system heeds the features array in hal/target.json. what build system are you using/ what command do you run to build an example application?

theotherjimmy, 2016-06-06:
super detailed discussion:
You are correct in identifying the problem: The macro DEVICE_SERIAL is not defined, leading this line https://github.com/mbedmicro/mbed/blob/master/hal/api/Serial.h#L21 to turn off the serial object. The reason that  https://github.com/mbedmicro/mbed/blob/master/hal/targets/hal/TARGET_STM/TARGET_STM32L1/TARGET_NUCLEO_L152RE/device.h is empty is that we have moved to a system that generates command line macros assignments from the ```hal/targets.json``` features array. The new location for these particular macros is here:https://github.com/mbedmicro/mbed/blob/master/hal/targets.json#L838 This means that your build system must generate, and pass, the macros previously defined in ```device.h```. 

If you exported a project before, you may have to export it again to get the command line style macros.

hesolium, 2016-06-06:
@theotherjimmy: "what build system are you using/ what command do you run to build an example application?"

I'm working under Ubuntu 14.04 LTS 64-bit (developing an application using Netbeans/Makefile). But it has nothing to do with error. With the previous revision (eg. 114) of mbed library everything worked properly. If the file 'device.h' is empty (no suitable #defines), then substantial part of the library code does not compile (because of conditional compilation).
What should be in the file 'device.h'? In the previous revisions (eg. 114) in this files where a lot of #define statements. In revision 121 these files are empty. Where is currently defined SERIAL_DEVICE variable?

hesolium, 2016-06-06:
@hesolium:"Where is currently defined SERIAL_DEVICE variable?"

Mistake! Should be DEVICE_SERIAL. In file 'Serial.h' compilation depends on this variable.

theotherjimmy, 2016-06-06:
> What should be in the file 'device.h'?

Just non-feature DEVICE_ defines. The current master looks correct to me.

> I'm working under Ubuntu 14.04 LTS 64-bit (developing an application using Netbeans/Makefile).

When did you last export your project?

>Where is currently defined SERIAL_DEVICE variable?

Well, that should be generated by the export.
For example, I generated a new makefile with the command ``` python workspace_tools/project.py -m NUCLEO_L152RE -i gcc_arm -p 0``` and the resulting makefile contains the [extra long] line 
```make
CC_FLAGS = $(CPU) -c -g -fno-common -fmessage-length=0 -Wall -Wextra -fno-exceptions -ffunction-sections -fdata-sections -fomit-frame-pointer -MMD -MP
CC_SYMBOLS = -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -DDEVICE_PORTIN=1 -DDEVICE_RTC=1 -DTOOLCHAIN_GCC -DTARGET_LIKE_CORTEX_M3 -DTARGET_CORTEX_M -DARM_MATH_CM3 -DDEVICE_ANALOGOUT=1 -DTARGET_STM32L1 -DTARGET_M3 -DDEVICE_SERIAL=1 -DTARGET_STM32L152RE -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M3 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FF_ARDUINO -DTARGET_STM -DTARGET_NUCLEO_L152RE -DMBED_BUILD_TIMESTAMP=1465248918.13 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 
```
That is where the DEVICE_SERIAL is defined.

hesolium, 2016-06-06:
@theotherjimmy: "Well, that should be generated by the export."

Thank you for your help. I do not use the remote compiler. My application is too big and work with the remote compiler is too slow and cumbersome. But you are right. Exported application (and 'device.h') is OK because makefile define all this variables. I have to modify my makefile.

theotherjimmy, 2016-06-06:
> I have to modify my makefile.

Sorry to hear that. You can generate a new makefile with the tools in the workspace_tools directory for guidance on what needs to be modified. It should just be appending the ```-DDEVICE_<feature>=1" lines to CC_SYMBOLS.

0xc0170, 2016-06-07:
@hasselmm Please look at https://github.com/mbedmicro/mbed/pull/1830,  and update your build system. Or as @theotherjimmy mentioned above.

I'll close this issue as it should be resolved, if not, please reopen

closed 2016-06-07 by 0xc0170