ARMmbed/mbed-os #1864: project.py returns a WindowsError instead of showing that the library is not found

adustm, 2016-06-07:
I launched the following command without building rtos and ethernet libraries prior to it:
`python ..\mbed\workspace_tools\project.py -c -m NUCLEO_F746ZG -p 96 -i uvision5`

This is failing with the following errors
```
Traceback (most recent call last):
  File "..\mbed\workspace_tools\project.py", line 176, in <module>
    setup_user_prj(project_dir, test.source_dir, test.dependencies)
  File "C:\mbed\workspace_tools\export\__init__.py", line 170, in setup_user_prj
    copy_tree(lib_path, join(user_lib, basename(lib_path)))
  File "C:\mbed\workspace_tools\export\__init__.py", line 152, in copy_tree
    copytree(src, dst)
  File "C:\Python27\lib\shutil.py", line 171, in copytree
    names = os.listdir(src)
WindowsError: [Error 3] The system cannot find the path specified: 'C:\\mbed\\.build\\rtos/*.*'
```

Martin asked me to create the issue, mentioning that I should not get a Windows Error but the script should return an error.
Kind regards

0xc0170, 2016-06-07:
cc @bridadan

bridadan, 2016-06-07:
Addressed in #1868

adustm, 2016-06-07:
Thanks @briadan

adustm, 2016-09-12:
Hello, I confirm that PR#2166 has fixed the issue
```
python tools/project.py -c -m NUCLEO_F746ZG -p 96 -i uvision5
gives:
Failed:
  * NUCLEO_F746ZG::uvision5     Library path 'C:\MCDGitW601\mbed\build\rtos/*.*' does not exist. Ensure that the library is built.
```

I am closing the issue
Kind regards
Armelle

closed 2016-09-12 by adustm