ARMmbed/mbed-os #1870: Exporter should error when libraries built with the correct toolchain are not available

bridadan, 2016-06-07:
To reproduce:

```
python workspace_tools/build.py -m K64F -t GCC_ARM --rtos
python workspace_tools/project.py -m K64F -i uvision5 -n RTOS_1
```

You will see that there is no built RTOS library included, just header files. If I repeat with the following:

```
python workspace_tools/build.py -m K64F -t ARM --rtos
python workspace_tools/project.py -m K64F -i uvision5 -n RTOS_1
```

The built RTOS library is included along with the header files.

In the first case, I would expect this to error with something along the lines of:

```
Library 'RTOS' was not found, please build with toolchain 'ARM'
```

OR it should just build it for you :)

theotherjimmy, 2016-06-15:
The example ```python workspace_tools/build.py -m K64F -t GCC_ARM --rtos``` works as intended, building the RTOS and RTX for me. Is this still an issue?

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-