ARMmbed/mbed-os #1947: .mbedignore does not recognise some paths correctly

LiyouZhou, 2016-06-15:
`./mbed-os/` is not recognised, you have to use `mbed-os/`
`*/TESTS/*` does not filter `./TESTS/*` you have to use `TESTS/*`

theotherjimmy, 2016-06-15:
The reason for this is that the current path is prepended to the pattern making your pattern ```some/path/here/*/TESTS/*``` which would not match ```some/path/here/TESTS/```.
Code reference https://github.com/mbedmicro/mbed/blob/master/tools/toolchains/__init__.py#L433

LiyouZhou, 2016-06-16:
Great, now that @theotherjimmy have isolated the problem, it should be quick to fix.

sg-, 2016-06-23:
@SeppoTakalo

SeppoTakalo, 2016-08-29:
See: https://github.com/ARMmbed/mbed-os/blob/master/docs/ignoring_files_from_build.md#syntax Rule 2.

`.mbedignore` does not use globbing. Assumption here is not correct therefore the reported issue is not valid.
If we would like it to use globbing, it is a new feature request.

sg-, 2016-08-29:
Thanks @SeppoTakalo

closed 2016-08-29 by sg-