ARMmbed/mbed-os #1953: uARM fail with compiled rtos lib

svastm, 2016-06-15:
The RTOS tests are failing on at least the NUCLEO_L476RG (uARM only).
```
Looking for NUCLEO_L476RG with MBEDLS
...port ready!
HOST: Reset target...
HOST: Property 'timeout' = '20'
HOST: Property 'host_test_name' = 'default_auto'
HOST: Property 'description' = 'Queue messaging'
HOST: Property 'test_id' = 'RTOS_5'
HOST: Start test...
OS ran out of threads!
```
I have launched some tests before the changes in the compilation flags and they were ok.

I tried to export a test to uvision and I had the same error with the compiled version of the rtos lib (such as rtos.ar and rtx.ar). But if i use the .c files and I compile the rtos lib inside the uvision project, the example works well.

0xc0170, 2016-06-15:
Can you share the commands please , and SHA of before the changes to test it locally please?

svastm, 2016-06-16:
The changes are in PR #1907 (`3db2c030d24782f8cab70ebf077a7b58d5f00973`) in the `arm.py` file.
All the targets are defined in *single thread mode* for the uARM toolchain.

0xc0170, 2016-06-16:
Thanks that explains it. uLib for ARMCC is not thread safe, therefore that change. The tests should take into account this.

cc @c1728p9

jamike, 2016-06-20:
The RTOS are also failing with  GCC_ARM , 
- as in file targets.json , ST config is set to "default_build": "small", the tools/toolchains/gcc.py is configured with  :
self.cc += ["-DMBED_RTOS_SINGLE_THREAD"]
self.cppc += ["-DMBED_RTOS_SINGLE_THREAD"]

svastm, 2016-07-01:
The RTOS tests are now marked as `NOT_SUPPORTED`.

closed 2016-07-01 by svastm