ARMmbed/mbed-os #1965: RTOS compile warnings

supamas, 2016-06-16:
This is my first issue submitted on github.  Sorry, I'm still learning.

I added mbed-rtos library to my already existing project.  Trying to compile without any call to the library produces these errors:

`Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 112, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 112, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 151, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 151, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 214, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 214, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 253, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 253, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 280, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 280, Col: 8
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_MemBox.c", Line: 108, Col: 27
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_MemBox.c", Line: 112, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_MemBox.c", Line: 160, Col: 39
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_MemBox.c", Line: 161, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
[... 11 lines cut ...]
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_System.c", Line: 245, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_System.c", Line: 245, Col: 6
[... cut ...]
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Success: Success! in "<a href="#" onClick="mbed_navigate_to('/ADC_SDcard_serial', function() { theitf.controls.get('sidebar_tabs').activate(1); }); return false;">Build Details</a>"
`

I apologize for the format.  Not sure if I'm doing this right.

bridadan, 2016-06-16:
Hi @supamas, thanks for reporting the issue! I've pasted your report down below in a more readable format. (By the way, I usually refer to the following guide when I check my formatting, hope it helps you out too! https://guides.github.com/features/mastering-markdown)

```
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 112, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 112, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 151, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 151, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 214, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 214, Col: 6
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 253, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 253, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 280, Col: 8
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_Mailbox.c", Line: 280, Col: 8
Warning: Incompatible redefinition of macro "__MBED_CMSIS_RTOS_CM" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 60, Col: 10
Warning: Incompatible redefinition of macro "__CMSIS_RTOS" in "mbed-rtos/rtx/TARGET_CORTEX_M/cmsis_os.h", Line: 62, Col: 10
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
[... cut ...]
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 209, Col: 17
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 212, Col: 13
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 214, Col: 18
Warning: Intrinsic is deprecated in "mbed-rtos/rtx/TARGET_CORTEX_M/rt_HAL_CM.h", Line: 216, Col: 13
Success: Success! in "<a href="#" onClick="mbed_navigate_to('/ADC_SDcard_serial', function() { theitf.controls.get('sidebar_tabs').activate(1); }); return false;">Build Details</a>"
```

neilt6, 2016-06-21:
The `Incompatible redefinition of macro` warnings were fixed a few weeks ago, just waiting for the updates to make their way back to the mbed site. The `Intrinsic is deprecated` warnings have been around for a while, and can be safely ignored. They need to be suppressed, but I have no idea when that is going to happen.

0xc0170, 2016-07-19:
Resolved via #2178

closed 2016-07-19 by 0xc0170