ARMmbed/mbed-os #1982: Exporters - uvision4 failed to build Nucleo F411RE project

toyowata, 2016-06-22:
Reported from a developer:

---
I have tried to build exported rtos uVision4 project from mbed classic using Template:

    New program -> NUCLEO-F411RE platform ->  Nucleo_rtos_basic template

Building the project on the uVision had been failed with three errors:

>    .\build\Nucleo_rtos_basic_copy.axf: Error: L6218E: Undefined symbol
>    __rt_lib_init (referred from rtx_conf_cm.o).
>    .\build\Nucleo_rtos_basic_copy.axf: Error: L6218E: Undefined symbol
>    __libspace_start (referred from rtx_conf_cm.o).
>    .\build\Nucleo_rtos_basic_copy.axf: Error: L6218E: Undefined symbol
>    __use_two_region_memory (referred from rtx_conf_cm.o).

I found that this is because the options of the project don't have the 'Use MicroLIB' option. Please refer to attached image.

[image]

I think all the templates should be reviewed.

0xc0170, 2016-06-22:
Thanks for reporting. There is option ``--microlib`` that should do the trick. I'll have a look at this

0xc0170, 2016-06-28:
I tested it on the latest master (mbed and project generator).

command:
```
python tools\project.py -m NUCLEO_F411RE -i uvision -n MBED_10 -b
```

uvision misc options:
```
-D__MICROLIB --restrict --multibyte_chars --apcs=interwork -c --library_type=microlib --no_vla -Otime --brief_diagnostics -O3 --cpu=Cortex-M4.fp -D__ASSERT_MSG --no_depend_system_headers --preinclude=mbed_config.h --md --split_sections --no_rtti --gnu
```

When I build without -b, there's an error about missing LAST/FIRST, I'll investigate.

This issue is for online IDE, thus when reported, should be tested compared to the latest mbed lib release and master (=to find out if we haven't yet fixed it).

0xc0170, 2016-06-28:
> When I build without -b, there's an error about missing LAST/FIRST, I'll investigate.

There's another issue that needs fixing, but not connected to microlib. I'll close this one, as it is resolved within this repository.

closed 2016-06-28 by 0xc0170