ARMmbed/mbed-os #1989: build_release.py --toolchain needs help text

BlackstoneEngineering, 2016-06-22:
## Problem
I want to use a toolchain, how do I know what the magic word is for the toolchain of my choice? How do I know what toolchains are supported? 

Running build_release.py --help nets the following
```
 ./build_release.py --help                                                                                                       1 ↵
[WARNING] Using default settings. Define your settings in the file "./mbed_settings.py"
Usage: build_release.py [options]

Options:
  -h, --help            show this help message and exit
  -o, --official        Build using only the official toolchain for each
                        target
  -j JOBS, --jobs=JOBS  Number of concurrent jobs (default 1). Use 0 for auto
                        based on host machine's number of CPUs
  -v, --verbose         Verbose diagnostic output
  -t TOOLCHAINS, --toolchains=TOOLCHAINS
                        Use toolchains names separated by comma
  -p PLATFORMS, --platforms=PLATFORMS
                        Build only for the platform namesseparated by comma
  -L, --list-config     List the platforms and toolchains in the release in
                        JSON
  --report-build=REPORT_BUILD_FILE_NAME
                        Output the build results to an junit xml file
  --build-tests=BUILD_TESTS
                        Build all tests in the given directories (relative to
                        /libraries/tests)
```

I have no clue what the options for `-t` are

## Solution
Either add tab completion for sub options, or add the currently supported toolchains to the help menu. In fact all of these commands would benefit from a example command in their help text.

BlackstoneEngineering, 2016-06-22:
Will probably be resolved by https://github.com/mbedmicro/mbed/pull/1976

theotherjimmy, 2016-06-22:
Nah, the PR that builds on #1976 will do it.

BlackstoneEngineering, 2016-06-22:
Would still love tab completion and additons to the help menu's in the future tho!

sg-, 2017-01-16:
I think this was reverted and not sure if re-applied. @theotherjimmy can you confirm if this is resolved?

theotherjimmy, 2017-01-17:
I don't think that build_release was ever updated. @sg- This is not yet resolved.

closed 2017-10-27 by ChiefBureaucraticOfficer