ARMmbed/mbed-os #1990: [build.py] debug-info prevents compilation for GCC_ARM

geky, 2016-06-22:
Ran the following:
``` bash
$ PYTHONPATH=. python tools/build.py -t GCC_ARM -m K64F
[WARNING] Using default settings. Define your settings in the file "./mbed_settings.py"
Building library CMSIS (K64F, GCC_ARM)
Copy: core_cm4_simd.h
Copy: arm_math.h
Copy: core_cm0plus.h
blablablabla
```
Works just fine. 

But oh no! Someone (not me I assure you) put a bug in my program. Time to debug.

Ran the following to add debug info:
```
$ PYTHONPATH=. python tools/build.py -t GCC_ARM -m K64F -o debug-info
[WARNING] Using default settings. Define your settings in the file "./mbed_settings.py"
GCC_ARM instance has no attribute 'notify_fun'

Completed in: (0.00)s

Build failures:
  * GCC_ARM::K64F
```

Here's some random info:
 - git log -n1 --oneline: f864b84 Merge pull request #1983 from mbedmicro/fix_config_tests 
 - python --version: Python 2.7.11+
 - cat /proc/version: Ubuntu 5.3.1-14ubuntu2.1

theotherjimmy, 2016-06-23:
That cat /proc/version is just the gcc version. it should look something like this:
```Linux version 3.13.0-66-generic (buildd@lgw01-26) (gcc version 4.8.2 (Ubuntu 4.8.2-19ubuntu1) ) #108-Ubuntu SMP Wed Oct 7 15:20:27 UTC 2015
```

screamerbg, 2016-06-24:
LGTM

PR #1995 (2016-06-24): [tools] Reorder option handling in toolchains
Options were parsed and logged before per-toolchain logging was setup.

Fixes #1990
commit 5a4c5df4 (2016-06-24): [tools] Reordered option handling in toolchains

Options were parsed and logged before per-toolchain
logging was setup.

Fixes #1990
closed 2016-06-24 by sg-