ARMmbed/mbed-os #1994: config file does not work when compiling tests

LiyouZhou, 2016-06-23:
cli version 0.7.13
mbed sha https://github.com/ARMmbed/mbed-os/#15b80ad7e9d069ba92cdeb0493015404bcd9ba31

To reproduce:
```
> mbed new myprog
> cd myprog
# edit mbed_app.json and add the following content:
{
    "custom_targets": {
        "K64F_SECURE": {
            "inherits": ["K64F"],
            "extra_labels_add":["K64F", "UVISOR_SUPPORTED"],
            "features_add": ["UVISOR"]
        }
    }
}

> mbed compile -t GCC_ARM -m K64F_SECURE 
# this works

> mbed compile -t GCC_ARM -m K64F_SECURE --tests
# Fails
Traceback (most recent call last):
  File "/tmp/myprog/mbed-os/core/tools/test.py", line 135, in <module>
    target = TARGET_MAP[options.mcu]
KeyError: 'K64F_SECURE'
[ERROR] 'K64F_SECURE'
[mbed ERROR] "python" returned error code 1.
[mbed ERROR] Command "python -u /tmp/myprog/mbed-os/core/tools/test.py -t GCC_ARM -m K64F_SECURE --source /tmp/myprog --build /tmp/myprog/.build/tests/K64F_SECURE/GCC_ARM --test-spec /tmp/myprog/.build/tests/K64F_SECURE/GCC_ARM/test_spec.json" in "/tmp/myprog"
---

>  mbed compile -t GCC_ARM -m K64F_SECURE --tests --config mbed_app.json
>  mbed test --compile -t GCC_ARM -m K64F_SECURE
>  mbed test --compile -t GCC_ARM -m K64F_SECURE --config mbed_app.json
# All of the above fails.
```

LiyouZhou, 2016-10-26:
@sg- @screamerbg @theotherjimmy I am encountering this again. Could I please get some help on this issue?

bridadan, 2016-10-26:
@LiyouZhou This was an issue in that release of mbed OS. However this should be resolved in any recent version of mbed OS. You'll also want to update your mbed CLI installation to the latest. This is the command you'll want to use:

```
mbed test --compile -t GCC_ARM -m K64F --app-config mbed_app.json
```

NOTE: If K64F_SECURE is a custom target, I believe this support was removed a while ago? Honestly not sure about it since I was never all that involved with it.

bridadan, 2016-11-02:
@LiyouZhou Are you still seeing this?

LiyouZhou, 2016-11-03:
@bridadan I am using --app-config mbed_app.json now, it work. Thanks.

bridadan, 2016-11-03:
Good to hear! Closing this.

closed 2016-11-03 by bridadan

hanmantg, 2017-11-30:
@LiyouZhou I am encountering the same issue after Compilation .Can  you please help to sort this issue(im using Windows platform).
[image]

theotherjimmy, 2017-11-30:
@hanmantg Can you open a new issue? commenting on long closed issues is not likely to garner much attention.

hanmantg, 2017-12-01:
@theotherjimmy Yep! i think but all those are solved except last Two error yet not sorted

theotherjimmy, 2017-12-01:
@hanmantg That's not the same issue. Please look at the lines above `[mbed] ERROR`, as those lines indicate that you encountered an error, not what the error was.