ARMmbed/mbed-os #1997: Option --source for get_config.py should be mandatory

theotherjimmy, 2016-06-23:
It errors out if you do not supply `--source`.
For example: 
```
$ python tools/get_config.py  -m K64F -t GCC_ARM -v 
Traceback (most recent call last):
  File "tools/get_config.py", line 62, in <module>
    params, macros, features = get_config(options.source_dir, target, toolchain)
  File "/home/jimbri01/src/mbedmicro/mbed/tools/build_api.py", line 85, in get_config
    src_paths = [src_paths[0]] + list(set(src_paths[1:]))
IndexError: list index out of range
```
should the `--source` argument be required, or should it default to "."?

bridadan, 2016-06-23:
cc @bogdanm

theotherjimmy, 2016-06-27:
or even dirname(dirname(__file__)).

theotherjimmy, 2016-12-13:
This has been in master for a very long time now

closed 2016-12-13 by theotherjimmy