ARMmbed/mbed-os #2004: build information as compile time defines

jupe, 2016-06-24:
Yotta had this feature and was useful for test applications we have: https://github.com/ARMmbed/yotta/issues/403

Propose is to collect build/module information like:

```
#define MBED_BUILD_INFO_HEADER    // to detect when these information is available
#define MBED_BUILD_TIME "YYYY-MM-DD HH:MM:SS"
#define MBED_BUILD_VCS_ID    <project_tag+sha>
#define MBED_BUILD_VCS_CLEAN   //only if git workarea is clean
#define MBED_BUILD_UUID     <unique build id>
#define MBED_BUILD_VCS_ID_<MODULE>  <module_tag+sha>
#define MBED_TOOLCHAIN  <used toolchain>
#define MBED_TARGET     <used target>
#define MBED_OS_VERSION <git tag/sha>
```

this was originally created to armmbed/mbed-cli: https://github.com/ARMmbed/mbed-cli/issues/182

jupe, 2016-10-27:
Any news about this @sg- ?

0xc0170, 2016-10-28:
``MBED_OS_VERSION`` - there's PR for this, #3096, can review it there.

The rest would be nice to have, +1 from me for having these.

@theotherjimmy @bridadan

bridadan, 2016-10-28:
Wouldn't this cause everything to be rebuilt every time due to the `#define MBED_BUILD_TIME`? Is this an option that would be off by default?

closed 2017-10-27 by ChiefBureaucraticOfficer