ARMmbed/mbed-os #2007: DigitalIn .is_connected() not working correctly

BlackstoneEngineering, 2016-06-24:
## Problem
The DigitalIn .is_connected() function serves to check if the pin has been connected, that is if the pin has been initialized with a pin value or is a no connection NC value. Currently the implementation is broken and is not populating correctly. It has something to do with how objects are getting passed around, I believe it is because the actual GPIO object is being initialized in the target code and not in the mbed hal code. I believe this is resulting in the value being initialized to 0 instead of to FFFF (which is the NC sentinal), thus resulting in incorrect behavior. 

## Solution
fix it so this function works. Also consider renaming function to `is_initialized()` as that makes more sense from a user perspective. 

## Example
In this example we create two DigitalIn objects, the first one `pin` is initialized and thus the `is_connected()` function should return true. The second object, `din` is not initialized, and thus should return false, however because of schenaniganz in the underlying library it does not actually work as expected and returns true. 
```cpp
#include "mbed.h"

DigitalIn pin(D4);
DigitalIn *din;
int main()
{
    pin.mode(PullNone); // PullUp, PullDown, PullNone, OpenDrain

    while(1) {
        wait(1);
        printf("din->is_connected() = %d \r\n",din->is_connected());
        printf("pin.is_connected() = %d \r\n",pin.is_connected());
        printf("\r\n");
    }
}
```

which results in the output of 
```
din->is_connected() = 1 
pin.is_connected() = 1 

din->is_connected() = 1 
pin.is_connected() = 1 

din->is_connected() = 1 
pin.is_connected() = 1 

din->is_connected() = 1 
pin.is_connected() = 1 
```

BlackstoneEngineering, 2016-06-24:
@sg- over here

BlackstoneEngineering, 2016-06-24:
This also effects DigitalOut and DigitalInOut

closed 2017-10-27 by ChiefBureaucraticOfficer

sg-, 2017-10-28:
Did we fix this?