ARMmbed/mbed-os #2017: No pwmout desctructor

LMESTM, 2016-06-27:
In hal/hal/pwmout_api.h there is the definition pwmout_free for targets to implement.
  void pwmout_free(pwmout_t *obj);
Many targets actually do have an implementation of this function, but this is never call from PwmOut class as there is no ~PwmOut.
Note: This is also valid for other objects, but I trying to test this one.

LMESTM, 2016-06-27:
I have a proposal in a branch that I tested ok on STM32 
https://github.com/LMESTM/mbed/commit/7a64b231054b027420b5ca9dfa7934635d82685b
but I amnot sure it's the proper implementation - so feedback would be welcome before  a possible PR is sent.  If not the proper way, I'd prefer mbed staff to address the issue.

seppestas, 2017-04-04:
This issue is also part / a duplicate of #3106.

LMESTM, 2017-05-23:
#3106 seems to be much more complete, so I can close this one and let's just track #3106

closed 2017-05-23 by LMESTM