ARMmbed/mbed-os #2024: No recovery mechanism from Thread instantiation faillure

pan-, 2016-06-27:
Instantiating an object of type [`Thread`](https://github.com/mbedmicro/mbed/blob/master/rtos/rtos/Thread.h) will start the thread immediately. 

If the thread can't be started then the system will enter in [`error` state](https://github.com/mbedmicro/mbed/blob/master/rtos/rtos/Thread.cpp#L52-L59). 

It should be possible to use the c++ `Thread` wrapper and recover from errors if the thread can't be started. 

One solution could be to not start the thread at instantiation point but instead let the user start  it explicitly by calling the function `Thread::start`. 

Another solution would be to embed a status in the `Thread` object which allow to verify if the thread is started or not after its instantiation.

```c++
class Thread { 

    bool is_running();
   
    operator bool() { 
        return is_running();
    }
};
```

This change preserve backward compatibility and let the user write simple code like this: 

```c++
Thread my_thread(do_something);
if (!my_thread) { 
   // thread not started
   // recovery here
}
```

0xc0170, 2016-06-28:
Thread object defines a method:

```
    /** State of this Thread
      @return  the State of this Thread
    */
    State get_state();
```

If an error occurs during ctor, I would expect to get a state, and check for ``Inactive`` - not created or terminated?

pan-, 2016-06-28:
@0xc0170 Thanks Martin, in that case then the instantiation should not enter the error state in case of error but instead set a state: https://github.com/mbedmicro/mbed/blob/master/rtos/rtos/Thread.cpp#L51-L61 .

0xc0170, 2016-06-28:
cc @geky What do you think about this proposal, to remove "fatal" error() and let user to handle some use cases

geky, 2016-06-28:
There was a lengthy discussion about this over here (unfortunately private): https://github.com/ARMmbed/mbed-os/pull/79

I agree implicitly starting the thread in the contructor is a relatively terrible idea due to the lack of error reporting (but hindsight is 20x20). Providing the error state is a good workaround, but auto-starting in the constructor has other downsides such as encouraging pre-main thread spawing, which can put your application in really weird states.

We decided the way forward was @pan-'s first suggestion. Provide a non-starting constructor ([here](https://github.com/mbedmicro/mbed/blob/master/rtos/rtos/Thread.h#L34)) and a Thread::start function ([here](https://github.com/mbedmicro/mbed/blob/master/rtos/rtos/Thread.h#L109)). This lets you catch errors in Thread::start's return value.

Unfortunately as @pan- points out, we can't simply remove the auto-starting constructor. The plan was to mark the constructor as deprecated once we had a mechanism to do so. Now we have MBED_DEPRECATED, so there's no reason we can't.

cc @niklas-arm: feel free to correct me if I missed anything

pan-, 2016-06-29:
Explicit deprecation of those constructors is a really good thing.

Only one way to start a thread is another good thing. 

I think I can close this now.

closed 2016-06-29 by pan-