ARMmbed/mbed-os #2040: Exporter - project.py contains sources + mbed lib object files (even when requested -b)

0xc0170, 2016-06-28:
To reproduce: 

```
python wtools\build.py -m KL05Z -t uARM
python tools\project.py -m KL05Z -i uvision -n MBED_10 -b
```

The project does not build as there are duplicated symbols

0xc0170, 2016-06-28:
the mbed folder (mbed lib built) and hal are present in the project file,they are even copied. So far I checked few places in the project/exporter.py scripts, not found yet the cause. I suspected  it would be either toolchain scan and copy or in the project, removal of mbed lib ``if not options.build``. 

cc @bridadan @screamerbg @theotherjimmy Any help appreciated to fix this regression

bridadan, 2016-06-28:
@0xc0170 I'm on this hash 986ed4f22ced11f5197fe2326af0f00e701896bc. I just tried this with the following commands:

```
python tools\build.py -m KL05Z -t uARM
python tools\project.py -m KL05Z -i uvision -n MBED_10 -b
```

And that actually worked fine.

When I exported without the `-b`, that failed with the following error when trying to compile:
```
mbed\TARGET_KL05Z\TOOLCHAIN_ARM_MICRO\MKL05Z4.sct(3): error: L6235E: More than one section matches selector - cannot all be FIRST/LAST.
```

Is this the behavior you mean?

0xc0170, 2016-06-28:
> And that actually worked fine.

You don't have mbed/ folders and hal/ . with ``-b`` , there sshould be only mbed folders, no hal

> Is this the behavior you mean?

Yes, exact error I was seeing

theotherjimmy, 2016-06-28:
I'm not seeing extra sources.
log:
```
$ python tools/project.py -m KL05Z -i uvision -n MBED_10 -b
<snip>
Successful exports:
  * KL05Z::uvision	/home/jimbri01/src/mbedmicro/mbed/.build/export/MBED_10_uvision_KL05Z.zip
$ unzip -l .build/export/MBED_10_uvision_KL05Z.zip "*.o" 
Archive:  .build/export/MBED_10_uvision_KL05Z.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
    27772  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/retarget.o
     5392  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/system_MKL05Z4.o
     4008  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/startup_MKL05Z4.o
     3360  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/sys.o
     4844  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/mbed_overrides.o
     6216  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/board.o
     4904  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/cmsis_nvic.o
---------                     -------
    56496                     7 files
$ unzip -l .build/export/MBED_10_uvision_KL05Z.zip "*.ar"
Archive:  .build/export/MBED_10_uvision_KL05Z.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
   496148  2016-06-28 11:40   MBED_10/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/mbed.ar
---------                     -------
   496148                     1 file
$ unzip -l .build/export/MBED_10_uvision_KL05Z.zip "*.c"
Archive:  .build/export/MBED_10_uvision_KL05Z.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
---------                     -------
        0                     0 files
$ unzip -l .build/export/MBED_10_uvision_KL05Z.zip "*.cpp"
Archive:  .build/export/MBED_10_uvision_KL05Z.zip
  Length      Date    Time    Name
---------  ---------- -----   ----
      233  2016-06-28 11:40   MBED_10/main.cpp
     2267  2016-06-28 11:40   MBED_10/env/test_env.cpp
---------                     -------
     2500                     2 files
$ ar t .build/export/workspace/MBED_10/lib/mbed/TARGET_KL05Z/TOOLCHAIN_ARM_MICRO/mbed.ar                 
PeripheralPins.o
gpio_api.o
analogin_api.o
gpio_irq_api.o
analogout_api.o
spi_api.o
serial_api.o
pinmap.o
port_api.o
sleep.o
pwmout_api.o
rtc_api.o
i2c_api.o
us_ticker.o
CAN.o
Ethernet.o
FileBase.o
BusIn.o
CallChain.o
AnalogIn.o
BusInOut.o
BusOut.o
FileLike.o
FilePath.o
FileSystemLike.o
I2CSlave.o
I2C.o
LocalFileSystem.o
InterruptIn.o
InterruptManager.o
RawSerial.o
SPI.o
SPISlave.o
Serial.o
SerialBase.o
Ticker.o
Stream.o
Timeout.o
assert.o
TimerEvent.o
Timer.o
critical.o
lp_ticker_api.o
error.o
gpio.o
mbed_interface.o
pinmap_common.o
rtc_time.o
semihost_api.o
us_ticker_api.o
ticker_api.o
wait_api.o

```

sg-, 2016-09-22:
Closing all older exporter related issues. Please reopen if still valid. https://developer.mbed.org/blog/entry/Fixed-exporters-in-the-mbed-Online-Compi/

closed 2016-09-22 by sg-