ARMmbed/mbed-os #2048: Features - casuing gcc arm to fail to export

0xc0170, 2016-06-28:
traceback with the command I used to get this failure:

```
mbed export -m K64F -i gcc_arm
Traceback (most recent call last):
  File "blinky\mbed-os\core\tools\project.py", line 213, in <module>
    tmp_path, report = export(project_dir, project_name, ide, mcu, project_dir[0], project_temp, clean=clean, make_zip=zip, extra_symbols=lib_symbols, sources_relative=sources_relative)
  File "blinky\mbed-os\core\tools\export\__init__.py", line 106, in export
    exporter.generate()
  File "blinky\mbed-os\core\tools\export\gccarm.py", line 128, in generate
    self.resources.win_to_unix()
  File "blinky\mbed-os\core\tools\toolchains\__init__.py", line 165, in win_to_unix
    self.features[f] = self.features[f].replace('\\', '/')
AttributeError: Resources instance has no attribute 'replace'
```

Using project.py is fine, no features probably triggered, thus it does not fail.

cc @screamerbg @bogdanm @theotherjimmy

theotherjimmy, 2016-06-28:
Thanks for reporting! I'll look into it for you.

theotherjimmy, 2016-06-28:
That should do it.

PR #2049 (2016-06-29): Fix a bug that caused the gcc_arm exporter to fail
Resolves #2048
commit feea8d59 (2016-06-29): Fix a bug that caused the gcc_arm exporter to fail

Resolves #2048
closed 2016-06-29 by sg-